jbuckmccready/CavalierContours #1: Add Implementation for Simultaneous Polyline Offsetting

jbuckmccready, 2020-01-18:
Add support for defining multiple polylines to apply offsets to and join offsets together. E.g. for use in defining islands inside of a Jordan curve and then applying repeated offsets.
Implementation would likely involve finding intersects between offset polylines and joining them together.

denisraison, 2020-04-05:
Hi @jbuckmccready, I was just about to ask that if it's possible to to use the polyline combine feature with the offset feature.

For example add multiple polylines shapes (clips or island) inside an outer shape and clip it using of of the modes (XOX, Exclude and etc) and then calling the offset tool with the result of that operation.

By the looks of it isn't possible just yet.

Bu the way I loved the concept of being able to use pline to offset and clipping arcs and line. It's a pain to use only points and approximation to convert it back to arcs and line components after offsetting it as we have to do with the clipping library.

jbuckmccready, 2020-04-05:
@denisraison Thanks for taking an interest in the project!

I actually just recently added the closed polyline combine (union, XOR, exclude, etc.), and hadn't given it much thought for the use in island offsetting and joining. Currently the combine returns two lists of closed polylines: one that represents remaining closed polylines, and the other that represents subtracted space (used in the case of islands after an exclude or union).

My original plan was to create a function that takes in an outer closed polyline and a list of inner closed polyline islands. Then I repeatedly offset the outer polyline inwards and the island polylines outwards, for each offset iteration I check for intersects between their offsets and slice and stitch them together, keeping only the slices that are no closer than an offset distance from either of the original polylines.

But your post has me thinking that what I described above is easily done using the exclude operation between the  outside offset polyline and the island offset polylines. Here's a quick sketch to show what I mean:
[image]
[image]

For multiple islands I think you would first union all of the island offset polylines, then exclude that result from the outside polyline offset.

The algorithm would be something like the following:

Let d = the offset delta per iteration.
1. Let A = offset of the outside polyline by d inwards, let B = each of the island polylines offset by d outwards.
2. Let C = union of all island offset polylines in B.
3. Exclude C from A, let D = polylines returned as remaining, let E = polylines returned as subtracted space.
4. Repeat step 1 by letting A = D offset by d inwards, and B = E offset by d outwards. Stop when A is empty after offset by d inwards.

Let me know if you have any other thoughts and/or ideas! I plan to get automated tests up and running soon with randomized fuzzing to catch numeric instabilities. I would also like to get a stabilized C API defined to make it easy for people to utilize the library from other languages.

jbuckmccready, 2020-04-05:
I created issues to track the [automated tests](https://github.com/jbuckmccready/CavalierContours/issues/4) and [C API](https://github.com/jbuckmccready/CavalierContours/issues/3). I will probably tackle those first before coming back to this issue.

denisraison, 2020-04-11:
Helloo @jbuckmccready 

That's exactly how I visualize the process being done and that's how we're currently using it with the clipper library even tho it does it slightly differently.

We feed the main "clipper" object with one closed subject/outer shape polyline and with a list of closed clips/islands/inner shapes and we then we apply a difference/exclude calculation.
With the result of the operation, we then apply the offsets we want.

I don't even think we would need to apply the 4th step you proposed there where you make A to be D. The input for the next iteration I think could still be the result from the first exclude but with the new offset value. But anyway, that doesn't really change anything so it doesn't really matter I think.

Thanks for getting back so quickly and with the great explanation.
I'm looking forward to seeing that implemented, that would be a gogo for me!!

If there is anything I can help the project with I'd be glad doing so.
Thanks

jbuckmccready, 2020-04-12:
You're right about not needing to let A = D for it to work and instead increase the offset delta. I was thinking it would help performance as the island polylines and outer closed polyline start to merge together, but that may not be the case - could implement it both ways and benchmark across various profiles!

I'm currently taking an initial stab at a stable C API I'm comfortable supporting going forward that allows the C++ code to change as I figure out exactly what containers/interface I want to expose. The C API also makes utilizing the library form other languages easier.

In terms of things to help with on the project:
- Let me know more about your use case for the library and if there are any other functions that would be useful. Is it for a c++ project or is the library being wrapped and called from another language?
- You could take a stab at implementing this function using the existing offset and combine functions, make a pull request/let me know how it goes (see the examples for how to use offset and combine functions, if there's not enough to go on from there let me know).
- I need to pick a c++ unit test library, one that ideally makes it easy to write randomized tests as well as typical discrete tests. I'm looking into [Catch2](https://github.com/catchorg/Catch2) since it seems simple and has generators for randomized tests. If you have any experience/suggestions that would be helpful.

jbuckmccready, 2020-04-25:
@denisraison I started working on the algorithm for this issue, the union all + exclude all is actually somewhat complex because an island can end up inside of the subtracted space from the union of two other islands.

The other issues are:
1. The way I have the code factored currently for combine operations makes a lot of copying/memory allocation necessary for the algorithm despite no island offsets overlapping and prevents the reuse of spatial indexes.
2. If there are many islands then some sort of spatial index on the island offsets is probably warranted to prevent the union all from turning into an N^2 problem.

The above two issues are just performance issues that can be improved once I get things tested, working, and committed.

Here's a sneak peak of the progress:
[image]

denisraison, 2020-04-26:
@jbuckmccready oh wow!!!!

Just responding to the previous post first...

- My use case for the library is mainly for offsetting any polyline that will most of the times, contain curves on it. The biggest struggle that I have to deal with the library I'm using now is how to be accurate when converting the extracted points back to simple components like straight lines and arcs... I feel like finding arcs from points for complex shapes can be sometimes more time consuming than offsetting itself.
- Our software is written in java where we're currently using JNI to wrap the c++ and using it as the bridge between the c++ lib and the Java code. That makes the internal c++ run faster than if we translate the code to Java.
- A great function and most need for us would be the ability to **link** the offsets with a line without going over an island. The function would find the nearest start point to the nearest offset. That would avoid the machine to go up and down to the next offset saving a lot of time... (if you are interested in this I'd be happy providing more details on it). I was thinking that the results of the offset would be provided in a tree-like structure. We didn't figure out yet how would be the best approach for it just yet...

I don't really think that performance would the biggest issue right now as just for the fact we don't need to translate the points back to arcs saves a whole amount of processing time required right now.

I just can't stop staring at the image you posted here lol, that's looking awesome and that's exactly what I would need the tool for.
I'm looking forward to seeing it committed and will implement it as soon as possible.

I saw you choose the google test library to create the tests and I particularly think that was a great option. And because trigonometry and c++ is not my strength, sorry, I didn't try to do it myself but I'm keen to help you the way I can even if it's just to evaluate the tool if I can.

By the way, I'm sorry for not responding to the previous post before sooner.

Thanks,

jbuckmccready, 2020-05-17:
@denisraison Hey, I've been pretty busy with other things, but here's an update on this.

I committed a working algorithm in [polylineoffsetislands.hpp](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polylineoffsetislands.hpp), but it is still somewhat a work in progress. I abandoned the combine approach as it became very complex in trying to track subtracted space polylines that arised due to unioning islands. 

Instead I take the following approach:

P = List of positive space closed polylines oriented counter clockwise (outer polylines)
N = List of negative space (island) closed polylines oriented clockwise
D = OffsetDelta

1. Offset P polylines inwards (+D, due to orientation)
2. Offset N polylines outwards (+D, due to orientation)
3. Create spatial index containing all of the offset polyline bounding boxes
4. Find all intersects between all offset polylines using the spatial index of all the bounding boxes and spatial indexes for each individual polyline's segments
5. Slice all offset polylines at the intersect points, discarding any slice that is closer than D distance from any of the original input polylines (before offset)
6. Stitch all slices together into closed polylines (they naturally stitch together end to start due to the initial clockwise/counter clockwise orientations)
7. Determine orientation of the resulting stitched closed polylines by finding their area, all clockwise oriented polylines act as negative space islands (N) for the next offset iteration, all counter clockwise polylines at as positive space (P) for the next offset iteration

So essentially the algorithm is just a modified version of the slice and stitch method I use for the basic offset algorithm.

The current implementation recomputes spatial indexes and performs extra allocations and copies that are not required. For step 6 it also just checks distance by comparing to all the other polylines (resulting in an O(N^2) complexity where N is the number of islands). Also it does not have consistent behavior around coincident/overlapping offset polylines. I also have plans for improving the existing simple offset algorithm, and this island offset algorithm, so they track information about how the output was created (e.g. from which polyline/segment index it was spawned from). I'll make issues to track my progress on these details of the algorithm.

NOTE: This algorithm properly handles islands within islands (which is what became difficult to solve using the combine operations as I currently define them), here's an example picture:
[image]

Here's another pretty picture I added to the readme:
[image]

denisraison, 2020-05-24:
Hey @jbuckmccready 

I've just implemented the feature here and I got tell you, this is brilliant.
I've done a shape with 7 islands inside it and it worked perfectly as expected.

That's great!! I'm looking forward to seeing the improvement that will allow us to track the polylines spawned.

I'll keep watching the commits and I'm happy to help wherever I can.
That's a great job @jbuckmccready congratulations for all the work done on getting it done.

Thanks,

jbuckmccready, 2020-05-29:
@denisraison That's great it's working for you.

I'm closing this issue and I made issues to track what I mentioned earlier in this thread. See the following issues: https://github.com/jbuckmccready/CavalierContours/issues/10 https://github.com/jbuckmccready/CavalierContours/issues/11

closed 2020-05-29 by jbuckmccready