jbuckmccready/CavalierContours #9: Bulge more than 1

denisraison, 2020-05-16:
Hello @jbuckmccready 

I've managed to implement the library with my application.
I have just hit a wall and would like to know if you have a solution for it.

I'm using some arcs that the bulge value ended up being more than 1, it's 1.25 for example.
And one of your checks is to check if the bulge is in between -1 and 1.

Do you have an idea on how to proceed with this?
Should I split the arc into 2, if so would you know the best way to do this?

Thanks,
Denis

denisraison, 2020-05-17:
Just another question as well.
What if I have a full circle and want to offset it using this library, how would that be done?

By the way, the library is amazing.
It's actually working much better than the clipper one, just by not having to convert the offset points bak to simple shapes.

Thanks for the effort and all the work done so far!!

jbuckmccready, 2020-05-17:
Yes, split it into two arcs, I would just use the midpoint of the arc sweep to perform the split.
Something like this in pseudocode:
```
IF arc.SweepAngle > PI THEN:
  midpoint = arc.ComputeMidPoint()
  halfSweepAngle = arc.SweepAngle / 2.0
  bulge = tan (halfSweepAngle / 4.0)
  vertex1 = { arc.StartPoint.X, arc.StartPoint.Y, bulge }
  vertex2 = { midpoint.X, midpoint.Y, bulge }
  vertex3 = { arc.EndPoint.X, arc.EndPoint.Y, 0.0 }
```
Where `vertex3` bulge would be non-zero if there was another arc that follows.

For a simple circle you can create it with just two vertexes that form two half circle arcs in a closed polyline. Here's a quick example:
```cpp
// circle parameters
double radius = 10.0;
double centerX = 1.0;
double centerY = 5.0;
bool isClockwise = true;

// bulge sign dictates direction/orientation
double bulge = isClockwise ? -1.0 : 1.0;

Polyline<double> circlePath;
// creating vertexes along the centerX line of the circle
circlePath.addVertex(centerX - radius, centerY, bulge);
circlePath.addVertex(centerX + radius, centerY, bulge);
circlePath.isClosed() = true;
```

denisraison, 2020-05-17:
Hi @jbuckmccready 

Do you know where I could find a formula for finding the midpoint of an arc?
The circle solution worked like a charm.

Thanks,

denisraison, 2020-05-17:
Hi @jbuckmccready 

That worked like a charm!! 

Thanks for the quick response it's really appreciated.

closed 2020-05-17 by denisraison