jbuckmccready/CavalierContours #13: Combining open polylines with dual offsets?

iisfaq, 2020-06-01:
Do you have any thoughts on implementing combining with open polylines?

In my case I need to have two open polylines with dual offsetts. I need to somehow combine the two dual offsets (I do not need to alter the original polyline)

Polyline #1 with dual offsets

[image]

Polyline #2 with dual offsets

[image]

Desired Output both polylines and dual offsets combined

[image]

Do you have any thoughts on this type of scanerio?

Chris

jbuckmccready, 2020-06-01:
Currently the union combine operation utilizes the the polylines being closed for orientation purposes (clockwise vs. counter clockwise which enables the slices to naturally stitch together end to start), and for determining which slices to discard (using the winding number).

For your particular use case assuming you have the original input lines you could just discard all slices which are closer to the original input lines than the offset delta given - that algorithm isn't implemented yet but it is just a variation of what all other functions currently do now. I might be able to add this at some point in the future - I don't have any expected time line to give for this though.

Or you could convert your dual offset open polylines into closed polylines by capping them off on the ends then performing the combine, and finally removing the segments that were added for the caps.

NOTE: Whenever I refer to a "slice" I am referring to a segment created by cutting a polyline at an intersection point.

iisfaq, 2020-06-02:
Thanks, I appreciate your comments.

I will see how I go with closing the poly lines into a closed polygon.

Cheers

Chris