jbuckmccready/CavalierContours #14: Offsetting multiple polylines without boundary curve

viggge, 2020-06-14:
Hi,
I would like to offset multiple different (but closed) polylines in one step.
It is similar to the example image given in the readme ([this one](https://raw.githubusercontent.com/jbuckmccready/CavalierContoursDoc/master/images/pretty_examples/islands_example1.png)) only that I don't want to specify the outer boundary curve. Hence, only the red dashed islands should be offsetted n steps outwards .
I think it should be doable but I couldn't figure it out (also not with the help of the CavalierContoursDev repo). 
So maybe you could give me an advice how to achieve this (if it's possible at all)?

jbuckmccready, 2020-06-14:
@viggge Hello,

Thanks for making the issue, the code was unnecessarily strict in requiring an outside counter clockwise boundary loop. I made a commit that allows for the "only islands" case you're interested in. See comiit https://github.com/jbuckmccready/CavalierContours/commit/79dc3d03336883b2eba9d0618ef45c0c08d94c9f.

I also updated the CavalierContoursDev repo to work correctly with no counter clockwise loops - comment out [this line](https://github.com/jbuckmccready/CavalierContoursDev/blob/master/interactiveui/plineoffsetislandsalgorithmview.cpp#L31) to try it.

I'll add the island offsetting to the [examples](https://github.com/jbuckmccready/CavalierContours/tree/master/examples) make it easier to follow, I'll probably get to that sometime in the next few days - or feel free to make a pull request with one if you end up using it.

jbuckmccready, 2020-07-07:
@viggge I updated the examples to include use of the the offset islands algoithm, see [here](https://github.com/jbuckmccready/CavalierContours/blob/master/examples/polylineoffsetislands.cpp).

jbuckmccready, 2020-07-21:
Closing this issue, open another if you find other issues with the island offsetting.

closed 2020-07-21 by jbuckmccready

viggge, 2020-07-25:
Sorry for the late reply, but it works perfectly!