jbuckmccready/CavalierContours #15: Give you another simple and fast curve equidistance algorithm

GridEyes-2010, 2020-06-21:


jbuckmccready, 2020-06-21:
@GridEyes-2010 Was this issue accidentally created? Not sure what this one is about.

GridEyes-2010, 2020-06-22:
I did it based on the distance field. First, I provided the unsigned version

GridEyes-2010, 2020-06-22:
#pragma once
#include <array>
#include <vector>
#include <map>
#include <list>
#include <iostream>
namespace FussenAlgo
{
	template<class TVert,class ftype>
	class MarchingSquarsCPU
	{
	public:
		//using ftype = float;
		//using TVert = std::array<ftype, 3>;
	protected:
		struct Edge {
			ftype ts[3];
			ftype te[3];
			 union {
				struct {
					int sid;
					int eid;
				};
				struct {
					int id[2];
				};
			};
		};
	public:

		void mscpuRoubst(std::vector<TVert>& pts, std::vector<ftype>& values, int udim, int vdim, ftype isoFace,
			std::vector<std::vector<TVert>>& curve, std::vector<bool>* property = nullptr)
		{
			std::vector<Edge> edges;
			edges.clear();
			int cudim = udim - 1;
			int cvdim = vdim - 1;
			edges.reserve(cudim*cvdim);
			for (int u = 0; u < cudim; ++u) {
				for (int v = 0; v < cvdim; ++v) {
					ftype v0 = values[idx(u, v, udim)];
					ftype v1 = values[idx(u + 1, v, udim)];
					ftype v2 = values[idx(u + 1, v + 1, udim)];
					ftype v3 = values[idx(u, v + 1, udim)];
					ftype p0[3], p1[3], p2[3], p3[3];
					p0[0] = pts[idx(u, v, udim)][0]; p1[0] = pts[idx(u + 1, v, udim)][0]; p2[0] = pts[idx(u + 1, v + 1, udim)][0]; p3[0] = pts[idx(u, v + 1, udim)][0];
					p0[1] = pts[idx(u, v, udim)][1]; p1[1] = pts[idx(u + 1, v, udim)][1]; p2[1] = pts[idx(u + 1, v + 1, udim)][1]; p3[1] = pts[idx(u, v + 1, udim)][1];
					p0[2] = pts[idx(u, v, udim)][2]; p1[2] = pts[idx(u + 1, v, udim)][2]; p2[2] = pts[idx(u + 1, v + 1, udim)][2]; p3[2] = pts[idx(u, v + 1, udim)][2];
					MachingKernel(v0, isoFace, v1, v2, v3, p0, p1, p3, edges, p2);
				}
			}
			sortEdges(curve, edges, property);
		}

		void mscpuRoubst(TVert*pts, ftype* values, int udim, int vdim, ftype isoFace,
			std::vector<std::vector<TVert>>& curve, std::vector<bool>* property = nullptr)
		{
			std::vector<Edge> edges;
			edges.clear();
			int cudim = udim - 1;
			int cvdim = vdim - 1;
			edges.reserve(cudim*cvdim);
			for (int u = 0; u < cudim; ++u) {
				for (int v = 0; v < cvdim; ++v) {
					ftype v0 = values[idx(u, v, udim)];
					ftype v1 = values[idx(u+1, v, udim)];
					ftype v2 = values[idx(u+1, v+1, udim)];
					ftype v3 = values[idx(u, v + 1, udim)];
					ftype p0[3], p1[3], p2[3], p3[3];
					p0[0] = pts[idx(u,v,udim)][0]; p1[0] = pts[idx(u+1,v,udim)][0]; p2[0] = pts[idx(u+1,v+1,udim)][0]; p3[0] = pts[idx(u,v+1,udim)][0];
[... 242 lines cut ...]
			std::sort(cpts.begin(), cpts.end());
			int unique = 0;

			//curve.resize(1);

			cpts[0].rindex = 0;
			TVert tp;
			tp[0] = cpts[0][0];
			tp[1] = cpts[0][1];
			tp[2] = cpts[0][2];
			curve[0].emplace_back(tp);
			for (int i = 1; i < npts; ++i) {
				if (cpts[i - 1] != cpts[i]) {
					++unique;
					tp[0] = cpts[i][0];
					tp[1] = cpts[i][1];
					tp[2] = cpts[i][2];
					curve[0].emplace_back(tp);
				}	
				cpts[i].rindex = unique;
			}
		}
	};

}

GridEyes-2010, 2020-06-22:
https://github.com/GridEyes-2010/offset-curve  (there is a demo)

jbuckmccready, 2020-06-22:
Thanks for sharing - from what I understand this algorithm works on rasterized images only (using the [marching squares](https://en.wikipedia.org/wiki/Marching_squares) approach), which is not very helpful for CAD/CAM applications that want to retain more of the geometric/topological information (e.g. arcs and lines).

jbuckmccready, 2020-07-21:
Closing this issue, while interesting (thanks for sharing!) I don't think it's directly applicable to this project.

closed 2020-07-21 by jbuckmccready