jbuckmccready/CavalierContours #16: Curves difference problem

lscandella, 2020-06-25:
Hello and compliments again for your work!

Recently one curves difference operation produced an unexpected result. According to the following image:

[image]

I was computing the difference between the white line and the red line. The result has been translated 100 units in the y direction and drawn in green. The problem is that all what is left of the red "circle" is missing.

I tried to simplify the problem before publishing here, but as soon as I make small variations in input data the problem  disappears...

That's why I'm attaching the raw data. Each line represents x, y, bulge of a vertex, the white line is on top, the red follows.

Thanks. L

563.775100244278 424.220150044019 0
576.860340670299 431.922000322341 -0.190218373487698
587.741076274678 433.633775050241 0
594.252489848897 432.178361231882 -0.18796247568131
603.294793423897 426.104475216784 0.110232815239728
651.216922681117 382.03851414704 -0.0137758363314688
651.920806164343 381.605531451554 0.108051353945198
666.922392086524 375.894478129712 -0.0820686709936642
671.577042425504 374.391866576656 0
690.326772987839 364.725000997945 0.100989586864969
698.52021110662 362.415604128012 0
701.285110378745 362.212197084576 -0.0805249311883016
705.954499347671 361.098487393067 0
745.56159576432 344.593186309873 -0.193821485444316
756.726145100352 345.581773314346 0
773.265853905451 340.563524507768 -0.273133458262753
787.856523016433 345.000213060298 0.0854677144777862
865.293982361801 339.761648185311 0.0130143782013616
844.750393065621 341.131435438289 0
820.878347715259 342.10000835824 -0.273598391484826
808.058970122095 350.401663325378 0
807.561967172104 351.399791287742 0.224295029889714
806.10932126658 352.544792761443 0.0379623113730011
804.880141864589 352.730993120664 0
771.005463169894 355.257710363912 -0.327249285592904
757.520623594488 366.77773970541 0.455225314296193
749.99212561602 370.746383022071 -0.271623459839902
735.011277905873 373.182396732415 0
719.668303868857 372.686094577875 -0.187566572970122
709.413658834857 376.296111803669 0
708.29977711401 377.252195515701 0.163342832864542
707.997473825034 377.374866980091 0
676.93278245763 379.292789107644 -0.132535668110463
669.527509773639 381.789601931447 0
667.356794441608 383.239034304494 -0.218493033341439
660.797189977336 393.89399369015 0
660.090463903118 399.676525428431 -0.2219870554849
664.026504726362 411.744565933748 0.0673180016113658
653.38422092449 417.130006696963 0
643.340296721685 420.615659002564 -0.156614660928744
636.009601183204 426.127902625301 0.116279710522012
614.205198006436 445.403873203374 0.248824877823294
584.004634641804 445.544695039742 0.0834463578326797
565.057827004872 431.15353751968 -0.0121681077925243
564.544948960788 430.634091127567 0.355940646831284

726.621074799823 374.75174833362 0
736.05375302248 374.008553753542 -0.193978199096541
738.062071964538 373.008967210915 0
743.846007511413 366.49065666404 -0.231482029859072
744.55133990716 363.950245251734 0
743.071237368098 356.002827038844 -0.0836239131013011
742.728609579215 355.067062639497 -1.38250253537883
725.172581353511 374.504915741864 -0.124340948711381

jbuckmccready, 2020-06-25:
@lscandella Thanks for the report, I'll look into it when I have some time to do so. Based on the fact that it goes away as soon as you make small changes one thing you can try right away is scaling up your input or adjusting the threshold values as it might be a numeric stability issue.

Try the following:
1. Scale your input up by 1000 (multiply all the vertex X and Y values by 1000) before performing the difference (if it doesn't go away immediately then try 10000). You can use the little helper function [here](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polyline.hpp#L68) for convenience if you're using c++. 
2. Play with increasing or decreasing by 1e-1 the threshold values used by the algorithms here: https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/mathutils.hpp
```cpp
// absolute threshold to be used for comparing reals generally
template <typename Real> constexpr Real realThreshold() { return Real(1e-8); }

// absolute threshold to be used for reals in common geometric computation (e.g. to check for
// singularities)
template <typename Real> constexpr Real realPrecision() { return Real(1e-5); }

// absolute threshold to be used for joining slices together at end points
template <typename Real> constexpr Real sliceJoinThreshold() { return Real(1e-4); }

// absolute threshold to be used for pruning invalid slices for offset
template <typename Real> constexpr Real offsetThreshold() { return Real(1e-4); }
```

If the issue goes away after scaling up the input and/or adjusting threshold values then it's likely a numeric stability issue. If the issue doesn't go away then it's more likely an actual bug in the code path. Let me know what you find.

jbuckmccready, 2020-06-25:
Alright there is definitely an issue in the code path because a debug assert is triggering when finding the intersects between the two curves (which is part of the process of findings the difference). I'll let you know what I find.

jbuckmccready, 2020-06-25:
For the second polyline's vertexes the second vertex from the bottom has a bulge of `-1.38250253537883`, bulge must always be between -1.0 and 1.0.

jbuckmccready, 2020-06-25:
See [this past issue](https://github.com/jbuckmccready/CavalierContours/issues/9) for how to deal with bulge greater than 1.0 and circles.

lscandella, 2020-06-26:
JBuck,

thanks, it works, 100% my fault. Thanks for your support!

I'm now closing this issue

closed 2020-06-26 by lscandella