jbuckmccready/CavalierContours #18: idea

grotius-cnc, 2020-07-10:
Are you interested in auto nesting with cavaliercontours input?

I modified this code : https://github.com/tamasmeszaros/libnest2d to work with Qt.
fixed a bug in the nfpplacer.hpp line 6. Its ready to integrate in the CadCam project now, but i like a change first.

What If we change the backend from clipper into cavaliercontours?

For this i made a ready to go QT project :
https://github.com/grotius-cnc/QT_Nest_with_clipper_backend/blob/master/README.md

the file particleswarm.hpp has some warnings. 

Let me know how you think about it.

jbuckmccready, 2020-07-10:
I haven't dug into any of the code but I am familiar with clipper. What parts of clipper are used by the nesting algorithm? Is it using offsetting or combine (AND/OR/NOT/XOR) operations for the nesting algorithm? I don't currently have a need for the nesting algorithm but seems like a cool project.

grotius-cnc, 2020-07-11:
HI,

At the moment i don't know much about the nesting code and clipper. It looks the directory "backend" is the clipper interface and contains 2 files. I think the developper needed clipper to show it's output somewhere?

For me it's a bad idea to start changing the nesting code input, before testing the current code output in cadcam for correctness.

On first sight it only uses clipper, points, paths, holes, but is see no (AND?OR?NOT or XOR) commands.

To get more familiar with the nesting code i will try to do some simple integration into cadcam today and try to do some autonesting. 

I have seen that it outputs object translation, rotation values for each product.

This is perfect and results in only use this 2 parameters at this moment to translate and rotate with the existing 3d_matrix.h in the cadcam lib directory. 

So the input could be a few important contour points, do the nesting. Get back the translation and rotation value, and move
the products including the cavalier pockets into their place. It can be a quite easy prodedure.

_but seems like a cool project._
Yes, a good nesting is a sort of holy grale for cnc software. I hope it works. Let you know later on.

grotius-cnc, 2020-07-15:
Hi, would let you know the solution for the nesting.

Made with my first Java Netbeans program for this. It was a struggle to find out, but now it's easy.
https://forum.linuxcnc.org/31-cad-cam/39555-nesting-tool-platform-independent#174769

Have a nice day !

jbuckmccready, 2020-07-21:
From what I can see you utilized a nesting algorithm that was written in Java (Nest4J) to make the app do what you needed, nice!

closed 2020-07-21 by jbuckmccready