jbuckmccready/CavalierContours #19: Have any plan to solve 3D Polyline offset？

YggSky, 2020-07-20:
current only 2d。

jbuckmccready, 2020-07-20:
I have no plans currently to add B-Spline support. I don't know what your use case is, but there are probably algorithms/apps/libraries to covert B splines into line + arc approximated paths that could then be fed into this library, and then the output could be used as is, or if needed there are likely algorithms/apps/libraries for approximating lines + arcs into B splines. In the future I might look into adding those approximation algorithms but currently I don't have any plans to do so.

If you're thinking of offsetting B Spline surfaces (not 2D paths) in 3D then the algorithm in this library is not applicable at all, it's entirely built up around offsetting paths in 2D space (finding distances, intersections, spatial indexing, etc. all works in 2D space and wouldn't work on a 3D surface).

YggSky, 2020-07-21:
> I have no plans currently to add B-Spline support. I don't know what your use case is, but there are probably algorithms/apps/libraries to covert B splines into line + arc approximated paths that [...]

thanks.by the way , [1] Liu, X.-Z., Yong, J.-H., Zheng, G.-Q., & Sun, J.-G. (2007). An offset algorithm for polyline curves. Computers in Industry, 58(3), 240–254. doi:10.1016/j.compind.2006.06.002    I search it on google ,but can't find direct  to read,all have special condition can do, of course I meet not .  how do you arrive literature and read,have some simple way?

jbuckmccready, 2020-07-21:
I usually end up using [sci-hub](https://en.wikipedia.org/wiki/Sci-Hub) for pay walled journals.
That paper is also just hosted in pdf form [here](https://hal.inria.fr/inria-00518005/document).

YggSky, 2020-07-22:
> I usually end up using [sci-hub](https://en.wikipedia.org/wiki/Sci-Hub) for pay walled journals.
> That paper is also just hosted in pdf form [here](https://hal.inria.fr/inria-00518005/document).

thanks a lot.

YggSky, 2020-07-24:
createFastApproxBoundingBox this function caculate arc  Approx AABB.

  double offsX = b * (v2.y() - v1.y()) / double(2);
  double offsY = -b * (v2.x() - v1.x()) / double(2);

I can't clear about it,i konw AABB,but can't understand,how could it can always equal to or bigger than the true bounding box.

jbuckmccready, 2020-07-24:
Check out this [page](http://www.lee-mac.com/bulgeconversion.html) to understand the relationship between the arc chord, bulge, and sagitta. Basically it's moving the arc chord line out along the sagitta towards the outside of the arc and then computing the min/max of all the end points. If you get the [CavalierContoursDev](https://github.com/jbuckmccready/CavalierContoursDev) project running you can visualize it by viewing the spatial index bounding boxes on an arc segment.

jbuckmccready, 2020-07-24:
Here's a quick sketch I made to visualize it:
[image]

The approximate AABB is found by just finding the min and max X and Y components of the P1, P2, P3, P4. Hope that helps explain it.

YggSky, 2020-07-27:
> Here's a quick sketch I made to visualize it:
> [image]
> 
> The approximate AABB is found by just finding the min and max X and Y components of the P1, P2, P3, P4. Hope that helps explain it.

it very clear，I see，thanks.

YggSky, 2020-08-03:
Is there any way to establish the relationship between the original segment(curve btween two PlineVertex) of Polyline and the offset segment of Polyline？ For some reason，for example ,if segment have color property，and is red, execute parallelOffset function, i hope to find offset segment corresponding to origin,so i can know what color it is. Of course ,offset segment compare to origin ,maybe start or end point coordenate change or this segment disappear or  add a new segment,new segment have a default color.

closed 2020-08-03 by YggSky