jbuckmccready/CavalierContours #20: Is there any way to establish the relationship between the original segment(curve btween two PlineVertex) of Polyline and the offset segment of Polyline？

YggSky, 2020-08-03:
(For some reason)for example ,if segment have color property，and is red, execute `cavc::parallelOffset` function, I hope to find offset segment corresponding to origin,so i can know the offset  segment and what color it is. Of course ,offset segment compare to origin ,maybe start or end point coordenate change or this segment disappear or  add a new segment,new segment have a default color.

jbuckmccready, 2020-08-03:
That's something I want to add, see issue https://github.com/jbuckmccready/CavalierContours/issues/11. I'm not sure when I'll get around to it.

YggSky, 2020-08-04:
> That's something I want to add, see issue #11. I'm not sure when I'll get around to it.

thanks,I know.