jbuckmccready/CavalierContours #22: Compiling with emscripten

yowzadave, 2020-08-19:
Hi! I’m wondering if this library should be able to compile to JS using emscripten? Thanks!

jbuckmccready, 2020-08-19:
I have never used emscripten but quickly looking at the project it looks like it creates JS from C/C++ compiled with clang so it should work. This library only depends on the standard library and a C++14 compatible compiler, and is continuously tested to compile on clang, gcc, and msvc.

yowzadave, 2020-08-20:
👍 sounds good! I'll give it a try. great looking library, by the way!

yowzadave, 2020-08-20:
I get an error on compilation:

>>> error: implicit conversion changes signedness: 'uint32_t' (aka 'unsigned int') to 'typename iterator_traits<__wrap_iter<PlineVertex<double> *>>::difference_type' (aka 'long') [-Werror,-Wsign-conversion]

jbuckmccready, 2020-08-20:
`-Werror` causes warnings to act as errors, remove that and it should compile. If you want to compile with `-Werror` then the implicit conversion signedness warning can be removed with an explicit `static_cast`. I'm surprised that there is a warning for that somewhere in the code, I usually compile with that warning on and add explicit `static_cast` when I can.

yowzadave, 2020-08-20:
I see--thanks. Turning off `-Wsign-conversion` and `-Wconversion` seems to allow it to compile successfully. It's flagging line 133 of "cavaliercontours.cpp" as doing the implicit conversion:
```
std::advance(start_it, start_index);
```

jbuckmccready, 2020-08-20:
Ah thanks for the feedback, I think `std::advance` function doesn't trigger the warning on gcc/msvc (which is typically what I'm compiling with during development). Was there other warnings as well? I'll maybe get around to eliminating them all on clang.

yowzadave, 2020-08-20:
That's the only one that I see!

jbuckmccready, 2020-11-23:
It's been a long time since you opened this issue but what OS (mac os, windows, or linux), standard library implementation (libstdc++ or libc++), and version of clang were you using when you compiled? I think the warning can be eliminated by casting the unsigned int to an Iter::difference_type (see [here](https://en.cppreference.com/w/cpp/iterator/iterator_traits)), but I am just surprised in all the ways I compile this project I haven't had the problem.

jbuckmccready, 2020-11-30:
I fixed it.

Commit: https://github.com/jbuckmccready/CavalierContours/commit/a38a84cba74f341d3c73803eff58a3c76afcc1c2

closed 2020-11-30 by jbuckmccready

rowanwins, 2020-12-16:
@yowzadave are you able to let us know how you got on with this is in the end - I'm also interested in making this algorithm available as a js library and using something like emscripten was something i considered.