jbuckmccready/CavalierContours #24: combinePolylines() in PlineCombineMode::Exclude mode with touching shapes can result in no output

fdarling, 2020-08-25:
I am working on integrating the CavalierContours into my project that demonstrates various geometric algorithm libraries and their ability to perform boolean operations on as well as offsetting of polygons/polylines. Here is a link to the project:

https://github.com/fdarling/line-arc-offset-demo

When running some tests I ran into a bug with the Exclude operation. If I subtract a vertical round-capped line from a horizontal rounded rectangle, it works quite nicely. However if I subtract the vertical round-capped line from the results (should do nothing, it's already been subtracted), it results in no geometry for the lower half of the vertical line.

Perhaps I am misunderstanding something in the API, but I don't think so because many other more complex test cases are succeeding. I'd appreciate any insight!

Pasted further down is a standalone C++ test case that illustrates the problem. It subtracts one Polyline from another (minuend - subtrahend), resulting in two Polylines. It then takes those resulting Polylines and subtracts the subtrahend from them once more, which should do nothing, but somehow it annihilates one of the original results.

Further down in the test case code I repeat the problematic subtraction with the Polylines hard-coded so that the issue can be narrowed down more easily. In other words, the second test is all that really needs to be debugged.

Here is an image showing the two shapes that I am subtracting. The minuend is in black, and the subrahend is in red:

[image]

Here is the console output of the test case:

```
---- Subtracting original Polylines ----
excludeResult.remaining.size():  2
excludeResult.subtracted.size(): 0
---- Subtracting resulting Polylines with original subrahend (shouldn't be able to cut anymore) ----
	excludeResult.remaining.size():  1
	excludeResult.subtracted.size(): 0
	----
	excludeResult.remaining.size():  0 (INCORRECT!)
	excludeResult.subtracted.size(): 0
	----
---- Subtracting problematic Polylines (should leave minuend untouched) ----
excludeResult.remaining.size():  0
excludeResult.subtracted.size(): 0
```

And the source code:

```cpp
#include <iostream>

#include "cavc/polylinecombine.hpp"

using namespace cavc;

#define KEEP_BULGES

#ifdef KEEP_BULGES
#define ADD_VERTEX(x, y, bulge) addVertex(x, y, bulge)
#else
#define ADD_VERTEX(x, y, bulge) addVertex(x, y, 0)
#endif

int main(int argc, char **argv)
{
    (void)argc;
    (void)argv;
    std::cout << "---- Subtracting original Polylines ----" << std::endl;
    {
        Polyline<double> a;
        a.ADD_VERTEX(-0.105, 0, 0);
        a.ADD_VERTEX(-0.105, 0.5, -1);
        a.ADD_VERTEX(-0.095, 0.5, 0);
        a.ADD_VERTEX(-0.095, 0, -1);
        a.isClosed() = true;

        Polyline<double> b;
        b.ADD_VERTEX(-0.25, 0.235, -0.414214);
        b.ADD_VERTEX(-0.255, 0.24, 0);
        b.ADD_VERTEX(-0.255, 0.29, -0.414214);
        b.ADD_VERTEX(-0.25, 0.295, 0);
        b.ADD_VERTEX(0.25, 0.295, -0.414214);
        b.ADD_VERTEX(0.255, 0.29, 0);
        b.ADD_VERTEX(0.255, 0.24, -0.414214);
        b.ADD_VERTEX(0.25, 0.235, 0);
        b.isClosed() = true;

        CombineResult<double> excludeResult = combinePolylines(a, b, PlineCombineMode::Exclude);
        std::cout << "excludeResult.remaining.size():  " << excludeResult.remaining.size() << std::endl;
        std::cout << "excludeResult.subtracted.size(): " << excludeResult.subtracted.size() << std::endl;
        std::cout << "---- Subtracting resulting Polylines with original subrahend (shouldn't be able to cut anymore) ----" << std::endl;
        for (std::vector< Polyline<double> >::const_iterator it = excludeResult.remaining.begin(); it != excludeResult.remaining.end(); ++it)
        {
            CombineResult<double> excludeResult2 = combinePolylines(*it, b, PlineCombineMode::Exclude);
            std::cout << "\texcludeResult.remaining.size():  " << excludeResult2.remaining.size();
            if (excludeResult2.remaining.size() != 1)
                std::cout << " (INCORRECT!)";
            std::cout << std::endl;
            std::cout << "\texcludeResult.subtracted.size(): " << excludeResult2.subtracted.size() << std::endl;
            std::cout << "\t----" << std::endl;
        }
    }
    std::cout << "---- Subtracting problematic Polylines (should leave minuend untouched) ----" << std::endl;
    {
        Polyline<double> a;
        a.ADD_VERTEX(-0.105, 0.235, 0);
        a.ADD_VERTEX(-0.095, 0.235, 0);
        a.ADD_VERTEX(-0.095, 0, -1);
        a.ADD_VERTEX(-0.105, 0, 0);
        a.isClosed() = true;

        Polyline<double> b;
        b.ADD_VERTEX(-0.25, 0.235, -0.414214);
        b.ADD_VERTEX(-0.255, 0.24, 0);
        b.ADD_VERTEX(-0.255, 0.29, -0.414214);
        b.ADD_VERTEX(-0.25, 0.295, 0);
        b.ADD_VERTEX(0.25, 0.295, -0.414214);
        b.ADD_VERTEX(0.255, 0.29, 0);
        b.ADD_VERTEX(0.255, 0.24, -0.414214);
        b.ADD_VERTEX(0.25, 0.235, 0);
        b.isClosed() = true;

        CombineResult<double> excludeResult = combinePolylines(a, b, PlineCombineMode::Exclude);
        std::cout << "excludeResult.remaining.size():  " << excludeResult.remaining.size() << std::endl;
        std::cout << "excludeResult.subtracted.size(): " << excludeResult.subtracted.size() << std::endl;
    }
    return 0;
}
```

jbuckmccready, 2020-08-25:
Thanks for the bug report and cool project! I found the problem and I should be able to commit a fix soon. The issue can arise any time the two shapes have only a coincident segment that they touch/overlap on.

jbuckmccready, 2020-08-30:
So the problem is little more difficult then I initially thought. Originally with this project I only set out to implement polyline offsetting but then I had an idea to use a similar slice and stitch approach for performing combine operations. Here's some general documentation regarding the current `combinePolylines` algorithm.

Here's an example initial input:
[image]
Closed polyline A in blue, closed polyline B in purple, intersects marked by red circles.

[image]
[image]
[image]
[image]

Some notes about the approach and assumptions:
1. Assumes input polylines are not self intersecting.
2. Assumes input polylines are closed (representing essentially 2D shapes).
3. Only supports two polylines at a time.
4. Slices are detected as being inside or outside of a polyline by computing the winding number of a point along the slice (specifically the mid point if the slice start and end are intersects, otherwise the mid point between the intersect and the next vertex along the slice's polyline). If the winding number is 0 then it is outside the polyline, if the winding number is 1 or -1 then it is inside the polyline.
5. Due to what was mentioned above in 4 there may be numerical stability/robustness issues if there are slices all layered very close together.

There is special case detection and handling in the case that the polylines do not intersect at all. There is also some special case handling in the case of coincident slices (e.g. given two polylines have the same shape an exclude operation will leave nothing remaining). But the special case handling for coincident slices is not fully built out and that is where the bug arises.
[image]
As you can see in the above image there are several possibilities that can arise in the case of coincident slices. Currently the code is already done/built out to fully detect all coincident slices (coincidence is "fuzzy" detected/approximated like any other floating point comparison). In order to fix this issue the code must be added to properly handle the cases, which may involve some additional geometric tests. Here's an example of what I mean:
A NOT B (A is blue, B is purple)
Case 1 (upper left): discard coincident slice for stitching.
Case 2 (upper right): include coincident slice for stitching.
Case 3 (lower left): discard coincident slices for stitching.
Case 4 (lower right): include bottom coincident slice, discard right coincident slice.

There are other cases for other operations (e.g. A AND B). The main thing of note is it appears that slices that are joined to the start and end points of a coincident slice may be tested to determine whether to keep or discard the coincident slice, such that a set of rules can be formed to handle all cases. E.g. in case 4 the bottom coincident slice is connected to two slices on B which are outside of A and two slices on A which are outside of B therefore it should be discarded, where as the right coincident slice is connected to one slice on B inside A and one slice on B outside A and one slice on A inside B and one slice on A outside B, therefore it should be included. My hypothesis is that these types of rules could be applied to all combine operation cases (union, exclude, intersect, xor) to cover all coincident input cases.

The above all seems very messy. It would be nice if there were other solutions! I'm still thinking on it, other ideas are welcome.

jbuckmccready, 2020-11-23:
Hey, I just made a commit that should fix the combinePolylines function to work in all coincident intersect cases. I added a few automated tests, including the polylines you posted in this issue. Create new issues if you find any more problems, the feedback is much appreciated!

It effectively uses what I described above but it's much simpler. Because of the constraints of the 2D topology and pruning of the slices I was able to modify the way it selects slices to be stitched together to cover all the cases. At some point I'll do a more detailed write up. I'd like to do the same for all the API function algorithms and put them in the project wiki.

Commit: https://github.com/jbuckmccready/CavalierContours/commit/c0b8ba04257ead894770f6dabd19872d4cd6b413

closed 2020-11-23 by jbuckmccready