jbuckmccready/CavalierContours #29: Support clipping open polylines

karl-nilsson, 2020-12-17:
Thanks for the excellent library!

Currently, boolean operations only support closed polylines:
https://github.com/jbuckmccready/CavalierContours/blob/b955785cc3db9689704d6135cbb60177fab835bb/include/cavc/polylinecombine.hpp#L415

It would be useful to support boolean operations between both open and closed polylines, as clipper does:
http://www.angusj.com/delphi/clipper/documentation/Docs/Units/ClipperLib/Types/ClipType.htm

jbuckmccready, 2020-12-17:
I think that would be fairly easy to implement based on how the algorithm works. I don't have an exact time line for this but I'll maybe take a look.

I'm actually in the process of rewriting the library in rust (with intent to create a clean C API wrapper, more elaborate documentation of the algorithms and code, guaranteed memory safety, and further performance optimizations around memory allocations). If I don't get around to supporting open polylines here then it will definitely be a target for the rust release.

I have some plans for supporting more around boolean operations. Specifically I want to implement support for generic shapes defined by a graph of positive and negative area polylines (negative areas represent holes) so that you can easily and efficiently do boolean operations between donut like shapes or even many layers deep of nested shapes (e.g. donuts within donuts). Right now to perform that type of operation you have to repeatedly perform boolean ops between only two polylines which is messy and inefficient.

karl-nilsson, 2020-12-17:
Will you continue to develop the c++ library?

jbuckmccready, 2020-12-17:
I'll fix any major bugs and accept pull requests for fixes/improvements. I don't know yet how much I will build out the API/features in C++. If I get frustrated/disillusioned with rust in the process of working on it then I may abandon it and stay with C++ (so far I'm really enjoying rust, so I don't see that as being likely but I have only rewritten the spatial index so far).

My hope/goal is to make the documentation of the algorithms themselves and quality of the code really good so that it becomes easy to port to any language. If I succeed in that goal then it may be easy for me/anyone to add any additions I make in rust to the C++.

I am also hoping to make an effective C API so it's accessible across the broadest range of languages via C FFI.

jbuckmccready, 2020-12-17:
For your particular request you could implement that behavior by finding the intersects between the open polyline and closed polyline, slice the open polyline at the intersect points and keep/discard (depending on the boolean op) any slices that are outside/inside of the closed polyline (can be tested using the winding number).

That is how I would implement it in the combine function.

ASHIT-AXAR, 2021-02-04:
@jbuckmccready
i'm leaving this here just in case of something other than "Rust"
www.dlang.org
it has worth a look. it's almost C with the same performance, OOP and modern programming, available for all major OS. you will find it very familiar.
also you can directly run C libraries without a change
just click "Show examples" below at home page and see how beautiful it is.

Nevsden, 2021-03-03:
> I think that would be fairly easy to implement based on how the algorithm works. I don't have an exact time line for this but I'll maybe take a look.
>
> I'm actually in the process of rewriting the [...]

Any update on the implementation of the rust portation? Do you have a repo for this and need people to help?

jbuckmccready, 2021-03-03:
@Nevsden The rust project is progressing and I plan to make the repo public sometime this week or beginning of next. There is a lot that is still under construction, but I'll make the repo available for people to contribute and see progress. My plan is to make more documentation within the repo on the project architecture and algorithms to facilitate contribution.

jbuckmccready, 2021-03-07:
@Nevsden The repository is now up [here](https://github.com/jbuckmccready/cavalier_contours) and under construction.

Nodobylove, 2025-08-07:
This is an amazing repo，I want to know Multiple combining polylines feature is done in Rust ？And Can give me some reference about the part of Boolean?

jbuckmccready, 2025-08-07:
> [@jbuckmccready](https://github.com/jbuckmccready) This is an amazing repo，I want to know Multiple combining polylines feature is done in Rust ？And Can give me some reference about the part of Boolean?

Boolean operation only supports between two closed (and not self intersecting) polylines currently (true for both Rust and c++ repository). The Rust repository has everything the c++ has and more.