jbuckmccready/CavalierContours #31: Multiple combining polylines

kadircanik, 2021-01-20:
Is there a way to make multi combining on polylines like clipper's AddPaths.

Initially i have 1 polyline, after offset i get multi polylines and i want to cut (cavc::PlineCombineMode::Exclude)
these polylines by more than one islands.

jbuckmccready, 2021-01-23:
It's not supported/implemented, but there is an algorithm to efficiently do it. I'm in the process of rewriting this library in Rust. See my replies to this issue: https://github.com/jbuckmccready/CavalierContours/issues/29

kadircanik, 2021-01-23:
This works

std::vector<cavc::Polyline<double>> GuidePolylines, aPolyList;
for (TopoDS_Wire aGuide : GuideWires2d)
{
	aGuide = CAM::ApproximateWireToLineAndArcs(aGuide, P.Tol);
	cavc::Polyline<double> aGuidePolyLine = WireToCavalier(aGuide, P.Tol);
	GuidePolylines = cavc::parallelOffset(aGuidePolyLine, P.GuideOffset);

	for (cavc::Polyline<double> aIsland : Islands) {
		aPolyList.clear();
		for (cavc::Polyline<double> aGuide : GuidePolylines) {
			cavc::CombineResult<double> res = cavc::combinePolylines(aGuide, aIsland, cavc::PlineCombineMode::Exclude);
			for (cavc::Polyline<double> aPoly : res.remaining)
				aPolyList.push_back(aPoly);
		}
		GuidePolylines = aPolyList;
	}
}

jbuckmccready, 2021-01-23:
I thought you needed a more generalized multi polyline combine operation - for your use case you may be able to just repeatedly call the combine function.

kadircanik, 2021-01-23:
Ofcourse it will be better, i have to go this way until you develop new solution :)