jbuckmccready/CavalierContours #32: Why not defined normalizing for zero vector? You can do it.

kadircanik, 2021-01-23:
auto lineVisitor = [&](PlineVertex<Real> const& v1, PlineVertex<Real> const& v2) {
				result.emplace_back();
				PlineOffsetSegment<Real>& seg = result.back();
				seg.collapsedArc = false;
				seg.origV2Pos = v2.pos();
				Vector2<Real> edge = v2.pos() - v1.pos();
				Vector2<Real> offsetV = offset * unitPerp(edge);
				seg.v1.pos() = v1.pos() + offsetV;
				seg.v1.bulge() = v1.bulge();
				seg.v2.pos() = v2.pos() + offsetV;
				seg.v2.bulge() = v2.bulge();
			};

v1 and v2 are same after offset. Then it says "normalize not defined for zero vector"

jbuckmccready, 2021-01-23:
Is there a particular issue/use case you're having problems with?

Based on the code you posted and the comment "normalize not defined for zero vector" (which is an assert message) I'm guessing you're trying to offset a polyline with repeated vertex positions.

You can remove all repeated vertex positions/singularities from a polyline before offsetting it by using the the "pruneSingularities" function, see [here](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polyline.hpp#L440).

kadircanik, 2021-01-23:
Initial polyline doesnt have any repeated vertex.
After first offset becomes repeated points.
So if i use pruneSingularities after every offset then no problem.

Actually my original data comes from splines.
I have an algorithm to approximate to arc and lines by using a tolerance value.
So i think there is problem about tangent continuity and that is the problem.

Thanks for your reply.

Best regards.

[image]

closed 2021-01-23 by kadircanik

jbuckmccready, 2021-01-23:
There shouldn't be any repeat vertex positions after using the offset function from this library. If that's the case (and it's not caused by some other code) please post an example polyline and offset value that has an output with a repeat position so I can take a look at it and possibly fix it.

kadircanik, 2021-01-23:
input.addVertex(-203.63194519,-96.94680216,0.02697234);
input.addVertex(-187.79469109,-116.72233086,0.01668655);
input.addVertex(-176.00833528,-129.07281375,0.02082326);
input.addVertex(-158.93552453,-144.46043948,0.02565229);
input.addVertex(-134.08367524,-162.97621244,0.03151757);
input.addVertex(-98.00733022,-183.91299360,0.03982232);
input.addVertex(-46.50695959,-204.70180207,0.02303935);
input.addVertex(-16.27645003,-212.68794228,0.02781530);
input.addVertex(18.18268709,-218.14635092,0.01594091);
input.addVertex(36.27370714,-219.39528227,0.01817670);
input.addVertex(55.18465348,-219.41134057,0.02106459);
input.addVertex(74.76349779,-217.89251560,0.02456213);
input.addVertex(94.79314488,-214.48206735,0.02600353);
input.addVertex(115.12294253,-208.85123101,0.02642203);
input.addVertex(135.72189141,-200.73456642,0.02585092);
input.addVertex(156.57724547,-189.87114031,0.02427264);
input.addVertex(177.69659406,-176.01128334,0.02180765);
input.addVertex(199.11756556,-158.93021455,0.01865154);
input.addVertex(220.92048783,-138.44625443,0.00786238);
input.addVertex(231.82592577,-127.08244089,0.00692832);
input.addVertex(242.87947421,-114.86019243,0.00595035);
input.addVertex(254.10966050,-101.78056529,0.00932004);
input.addVertex(277.61603953,-72.59387275,0.00811369);
input.addVertex(289.34059839,-56.95869863,0.01229424);
input.addVertex(300.71217329,-40.43714422,0.01739294);
input.addVertex(311.26052358,-22.98888267,0.02372059);
input.addVertex(320.42909563,-4.57284709,0.01455487);
input.addVertex(324.26716772,4.93401305,0.01661988);
input.addVertex(327.51677717,14.69436288,0.01880195);
input.addVertex(330.08114538,24.71334948,0.02101803);
input.addVertex(331.85593487,34.99613315,0.02313803);
input.addVertex(332.72878057,45.54788030,0.05098161);
input.addVertex(331.26882313,67.64092220,0.02375479);
input.addVertex(328.82047909,78.90742651,0.04541717);
input.addVertex(320.41987970,101.67920107,0.04489137);
input.addVertex(307.48050259,123.61771773,0.04805919);
input.addVertex(290.18072064,143.44168997,0.02629875);
input.addVertex(280.05988868,152.03134100,0.02860984);
input.addVertex(268.97140256,159.52148778,0.03137682);
input.addVertex(256.96324249,165.68529007,0.03412080);
input.addVertex(244.09163349,170.28142830,0.03250021);
input.addVertex(230.45599033,173.17455254,0.03023251);
input.addVertex(216.20938507,174.36906844,0.02819038);
input.addVertex(201.52385434,173.88317288,0.02656804);
input.addVertex(186.58938350,171.74168190,0.05084443);
input.addVertex(156.59706475,162.52936033,0.05199086);
input.addVertex(128.43963887,147.08762779,0.02871815);
input.addVertex(115.85012350,137.24201871,0.03179395);
input.addVertex(104.54939676,125.96036013,0.03530537);
input.addVertex(94.83654121,113.24019807,0.03891806);
input.addVertex(87.03547739,99.08265560,0.08725951);
input.addVertex(78.57542812,66.19600449,0.01674710);
input.addVertex(78.21428070,57.59105300,0.00849105);
input.addVertex(78.28642729,48.85270335,-0.00000000);
input.addVertex(78.44101992,43.01511118,-0.00351339);
input.addVertex(78.61959588,35.29073095,-0.01485956);
input.addVertex(78.50352528,25.23774871,-0.01446419);
input.addVertex(78.09825877,19.69222846,-0.02264627);
input.addVertex(77.19910416,13.59379641,-0.03470903);
input.addVertex(75.44679721,7.00852780,-0.05106935);
input.addVertex(72.26588065,0.09172374,-0.06782527);
input.addVertex(66.74785221,-6.85762899,-0.03167658);
input.addVertex(62.91914558,-10.08830470,-0.02904337);
input.addVertex(58.25802828,-13.13973232,-0.02709642);
input.addVertex(52.75293678,-15.91981789,-0.02608795);
input.addVertex(46.42396210,-18.31377986,-0.05617050);
input.addVertex(31.05761100,-21.38876114,-0.03268298);
input.addVertex(22.87859190,-21.55483916,-0.04089756);
input.addVertex(14.59905490,-20.51161541,-0.05523480);
input.addVertex(6.67325201,-17.92691673,-0.06416232);
input.addVertex(-0.42328941,-13.47447846,-0.05831626);
input.addVertex(-6.57381415,-7.03761944,-0.02438018);
input.addVertex(-9.25049904,-3.13463856,-0.02113304);
input.addVertex(-11.70298954,1.24882936,-0.01787519);
input.addVertex(-13.94107211,6.10151849,-0.01474219);
input.addVertex(-15.97969955,11.40504736,-0.01177918);
input.addVertex(-17.83984896,17.13274732,-0.00896603);
input.addVertex(-19.54948987,23.24834143,-0.00623672);
input.addVertex(-21.14467543,29.70445686,-0.00348807);
input.addVertex(-22.67077062,36.44095167,-0.00000000);
input.addVertex(-24.18383317,43.38303504,0.00728991);
input.addVertex(-27.49370307,57.77596166,0.01408427);
input.addVertex(-31.47664073,72.25729204,0.02295499);
input.addVertex(-36.58493495,86.61593790,0.01529504);
input.addVertex(-39.68934444,93.55490018,0.01824382);
input.addVertex(-43.29994769,100.36128217,0.04632528);
input.addVertex(-52.64169811,113.60126405,0.05470060);
input.addVertex(-65.55385298,125.66081227,0.02506112);
input.addVertex(-73.42094733,130.97486192,0.02399802);
input.addVertex(-82.23969631,135.74850854,0.04789011);
input.addVertex(-102.78734743,143.28681864,0.05185748);
input.addVertex(-126.05086300,146.90462939,0.02935117);
input.addVertex(-138.07740889,146.79686226,0.07002852);
input.addVertex(-162.56799908,141.68660242,0.03357111);
input.addVertex(-174.28756395,136.55644137,0.03216696);
input.addVertex(-185.62401577,129.69685510,0.03068359);
input.addVertex(-196.43427265,121.13550029,0.02939350);
input.addVertex(-206.55399063,110.92498005,0.02851861);
input.addVertex(-215.79497943,99.14708278,0.02824126);
input.addVertex(-223.94232973,85.91756827,0.05919729);
input.addVertex(-236.11294234,55.21519668,0.02864957);
input.addVertex(-239.53785344,38.52298694,0.02793999);
input.addVertex(-241.11189744,20.91788041,0.02722409);
input.addVertex(-240.72320185,2.52726941,0.02663635);
input.addVertex(-238.26050135,-16.48893635,0.02629476);
input.addVertex(-233.61419678,-35.93358725,0.05390305);
input.addVertex(-217.00076294,-75.73612213,0.02948234);

kadircanik, 2021-01-23:
Offset value is 2

jbuckmccready, 2021-01-23:
Thanks for the example but it doesn't seem to produce repeat positions?
```cpp
Polyline<double> input;
input.addVertex(-203.63194519,-96.94680216,0.02697234);
input.addVertex(-187.79469109,-116.72233086,0.01668655);
// ... add all the vertexes
Polyline<double> offsetResult = parallelOffset(input, 2.0)[0];
Polyline<double> pruned = pruneSingularities(m_inputPolyline, 1e-5);
assert(offsetResult.size() == pruned.size());
```
I would expect the sizes to be different after pruning to remove repeat vertexes?

kadircanik, 2021-01-23:
Yes, you are right. It was my mistake, sorry for distributing you.
Took my 2 hours to find the problem.

kadircanik, 2021-01-24:
----------Initial------------------
   0  -203.632   -96.947   0.02697
   1  -187.795  -116.722   0.01669
   2  -176.008  -129.073   0.02082
   3  -158.936  -144.460   0.02565
   4  -134.084  -162.976   0.03152
   5   -98.007  -183.913   0.03982
   6   -46.507  -204.702   0.02304
   7   -16.276  -212.688   0.02782
   8    18.183  -218.146   0.01594
   9    36.274  -219.395   0.01818
  10    55.185  -219.411   0.02106
  11    74.763  -217.893   0.02456
  12    94.793  -214.482   0.02600
  13   115.123  -208.851   0.02642
  14   135.722  -200.735   0.02585
  15   156.577  -189.871   0.02427
  16   177.697  -176.011   0.02181
  17   199.118  -158.930   0.01865
  18   220.920  -138.446   0.00786
  19   231.826  -127.082   0.00693
  20   242.879  -114.860   0.00595
  21   254.110  -101.781   0.00932
  22   277.616   -72.594   0.00811
  23   289.341   -56.959   0.01229
  24   300.712   -40.437   0.01739
  25   311.261   -22.989   0.02372
  26   320.429    -4.573   0.01455
  27   324.267     4.934   0.01662
  28   327.517    14.694   0.01880
  29   330.081    24.713   0.02102
  30   331.856    34.996   0.02314
  31   332.729    45.548   0.05098
  32   331.269    67.641   0.02375
  33   328.820    78.907   0.04542
  34   320.420   101.679   0.04489
  35   307.481   123.618   0.04806
  36   290.181   143.442   0.02630
  37   280.060   152.031   0.02861
  38   268.971   159.521   0.03138
  39   256.963   165.685   0.03412
  40   244.092   170.281   0.03250
  41   230.456   173.175   0.03023
  42   216.209   174.369   0.02819
  43   201.524   173.883   0.02657
  44   186.589   171.742   0.05084
  45   156.597   162.529   0.05199
  46   128.440   147.088   0.02872
  47   115.850   137.242   0.03179
  48   104.549   125.960   0.03531
  49    94.837   113.240   0.03892
  50    87.035    99.083   0.08726
  51    78.575    66.196   0.01675
  52    78.214    57.591   0.00849
  53    78.286    48.853  -0.00000
  54    78.441    43.015  -0.00351
  55    78.620    35.291  -0.01486
  56    78.504    25.238  -0.01446
  57    78.098    19.692  -0.02265
  58    77.199    13.594  -0.03471
  59    75.447     7.009  -0.05107
  60    72.266     0.092  -0.06783
  61    66.748    -6.858  -0.03168
  62    62.919   -10.088  -0.02904
  63    58.258   -13.140  -0.02710
  64    52.753   -15.920  -0.02609
  65    46.424   -18.314  -0.05617
  66    31.058   -21.389  -0.03268
  67    22.879   -21.555  -0.04090
  68    14.599   -20.512  -0.05523
[... 174 lines cut ...]
 134  -137.401   136.820  -0.00044
 135  -137.419   136.819   0.07003
 136  -159.181   132.278  -0.00001
 137  -159.181   132.278   0.03357
 138  -169.672   127.686   0.03217
 139  -179.908   121.492   0.03068
 140  -189.757   113.692  -0.00002
 141  -189.757   113.691   0.02939
 142  -199.050   104.315  -0.00004
 143  -199.051   104.314   0.02852
 144  -207.589    93.433  -0.00005
 145  -207.590    93.431   0.02824
 146  -215.145    81.163  -0.00003
 147  -215.146    81.161   0.05920
 148  -226.447    52.653  -0.00013
 149  -226.448    52.648   0.02865
 150  -229.643    37.077   0.02794
 151  -231.117    20.585   0.02722
 152  -230.752     3.282  -0.00001
 153  -230.752     3.282   0.02664
 154  -228.426   -14.678  -0.00002
 155  -228.426   -14.679   0.02629
 156  -224.024   -33.099   0.05390
 157  -208.241   -70.913   0.02948
 158  -195.501   -91.126  -0.00000

jbuckmccready, 2021-01-24:
I know there can be things that are very close to the threshold for being a singularity position, or line vs. arc in the bulge, but in testing the example you just gave it never fails to generate an offset (offset value 2) and the offset looks correct (at least down to the default threshold values).

What problem are you having? You may want to try to scaling up the inputs if you're running into problems around the thresholds, e.g. scale all X and Y value of the vertexes up by 1000, perform the offset (also scaling the offset value by 1000) and then scale all X and Y values of the output down by 1000 to get the final result. The scaling helps to prevent loss of precision due to the floats becoming very small in intermediate mathematical operations.

If you're trying to go from some higher order spline to approximate lines and arcs and then back to a higher order spline after performing the offset then you might have to add some error/threshold detection in reconstructing the spline since it wont necessarily be one to one with the approximation generated before the offset (e.g. in the case of collapsing arcs, or crossing threshold of the bulge from arc to line).

kadircanik, 2021-01-24:
Thank you for your reply.
Your prune function is good that kind of problems.
I had some other problems most of them are crashes.
It is interesting when i debug to find problem, i couldnt find.

for example now crash at that line, i is 0 but i get stack overflow.
inline Real const& operator[](std::size_t i) const { return m_data[i]; }

I tested clipper for 2 or 3 weeks and it looks more robust.
Never made me sad.

I decided to use clipper but i also need arcs.
So i will make a polyline classlike yours (using bulge)
I will use the clipper for offset and boolean operations.
Of course after every offset or boolean i will make arc fitting to save in my polyline class.

Your polyline class is template based, complex for me.
Do you know any simple polygon class?

jbuckmccready, 2021-01-24:
You mention having lots of problems but you don't provide any concrete examples, or enough information to be a reproducible example. I'd like to help but you aren't giving me enough information to work with.

I don't know how `inline Real const& operator[](std::size_t i) const { return m_data[i]; }` (which I'm guessing you copied from this line [here](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/vector.hpp#L35) can give you a stack overflow unless there is some gross memory corruption causing it to execute some other arbitrary code, or some recursive code around it (that is not part of this library) which is causing the overflow. If you have memory corruption caused by some other code in your program then when you go to use this library bad things could happen that are totally unrelated to this library. These seem like issues/bugs that arise around code using this library, but aren't actually bugs in this library.

I don't know of any simple polygon class, the templates used in this library are very simple and can always just be given a `double` or `float` and work.

If clipper resolves your issues that's good and I wish you the best of luck, but I don't think any of the problems you're having are related to bugs in this library, and I'd like to help if you can give me extracted examples that demonstrate problems in this library.

kadircanik, 2021-01-24:
Hi Jedidiah,

Is that possible to connect my computer using teamviewer etc?

Kadir Canik

Kimden: Jedidiah Buck McCready<mailto:notifications@github.com>
Gönderilme: 25 Ocak 2021 Pazartesi 02:17
Kime: jbuckmccready/CavalierContours<mailto:CavalierContours@noreply.github.com>
Bilgi: kadircanik<mailto:kadircanik@msn.com>; State change<mailto:state_change@noreply.github.com>
Konu: Re: [jbuckmccready/CavalierContours] Why not defined normalizing for zero vector? You can do it. (#32)

You mention having lots of problems but you don't provide any concrete examples, or enough information to be a reproducible example. I'd like to help but you aren't giving me enough information to work with.

I don't know how inline Real const& operator[](std::size_t i) const { return m_data[i]; } (which I'm guessing you copied from this line here<https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/vector.hpp#L35> can give you a stack overflow unless there is some gross memory corruption causing it to execute some other arbitrary code, or some recursive code around it (that is not part of this library) which is causing the overflow. If you have memory corruption caused by some other code in your program then when you go to use this library bad things could happen that are totally unrelated to this library. These seem like issues/bugs that arise around code using this library, but aren't actually bugs in this library.

I don't know of any simple polygon class, the templates used in this library are very simple and can always just be given a double or float and work.

If clipper resolves your issues that's good and I wish you the best of luck, but I don't think any of the problems you're having are related to bugs in this library, and I'd like to help if you can give me extracted examples that demonstrate problems in this library.

kadircanik, 2021-01-25:
[image]
[image]
[image]

First i have outermost polyline;
- Offset it 20 to inside
- Offset island 20 outside
- Cutout the island
- Then i have polyline which on first picture.
- Then offset 3 to inside
- Big polyline ofsetts to inside but small polyline ofsetss to outside???
- If i change offset value smaller than 3 no problem, offsets to inside

kadircanik, 2021-01-25:
I think i found a solution.
I always use guides as counterclockwise.
I see that if offset value is bigger than the polyline bounding box problem is starting but polyline is changing direction to
clockwise, so i can catch it.

Previus problem about crash is related this.

jbuckmccready, 2021-01-25:
Yes you can check for direction inversion (`(getArea(polyline) < 0) != (getArea(polylineOffset) < 0)`) to ensure it doesn't accidentally invert. Under normal cases it should catch that it becomes inverted by the distance checks it makes but if the input polyline is really small (nearly a point) then it can fail to catch it using just the distance checks.

In the example of the polyline in the lower right it's not very small so the inversion check shouldn't be needed, I am not sure why it's not catching the inversion, can you post the input polyline vertexes and offset value for that polyline in the lower right? It looks like it only has 3 vertexes.

kadircanik, 2021-01-25:
Wow, no crash anymore :)
Looks great now.
Your solutions is really great.

Small Polyline;

   0    80.01414   -39.99999   0.00000      0.00
   1    89.99999   -39.99999   0.00000      0.00
   2    89.99999   -30.01414  -0.41339    -89.84