jbuckmccready/CavalierContours #34: Not closed polyline abnormal with large offset

tomicfish, 2021-01-29:
cavc::Polyline<double> input;
//add points to input
for (size_t i = 0; i < PX.size(); i++)
{
	input.addVertex(PX[i], PY[i], 0);
}
input.isClosed() = false;
double offset = 150; 
std::vector<cavc::Polyline<double>> results = cavc::parallelOffset(input, offset);

When I set the offset to 3, the result is correct.
But when I set the offset greater than 20, I can't get the correct curve.
Here's the data I use.
[PX.txt](https://github.com/jbuckmccready/CavalierContours/files/5891967/PX.txt)
[PY.txt](https://github.com/jbuckmccready/CavalierContours/files/5891969/PY.txt)
Please give me some advice.

jbuckmccready, 2021-01-29:
I haven't tested the data you gave, but before I do have you tried scaling the inputs up by some factor (e.g. multiply all vertex x and y values and the offset value by1000) before performing he offset and then scaling the result back down? Might be a float threshold bottoming out problem.

tomicfish, 2021-01-29:
Hi jbuckmccready, 
I've verified it according to your suggestion, but the result still seems incorrect.
I multiply all vertex x and y values and the offset value by 100. I entered 3569 sets of data, but only got 1367 sets of data when the offset is set to 50. 
If I don't scale the inputs,  I entered 3569 sets of data, and got 4216 sets of data when the offset is set to 5.
If I don't scale the inputs,  I entered 3569 sets of data, and got 2316 sets of data when the offset is set to 20.
The inputs is actually the coordinates of the edge points of an object in a image.
Therefore, I want to get a parallel curve of the edge, and I can also confirm that the curve with offset of 20 is still in the valid range of the image，but I don't understand why fewer results have been obtained.

jbuckmccready, 2021-01-29:
Have you actually visualized the result to confirm it's wrong? The number of vertexes returned may be larger or smaller than the number of input vertexes, that's not indicative of a problem. Vertexes are added when offsetting outward around a corner to produce an arc to maintain a constant distance, and vertexes are removed when arcs collapse or sub regions collapse. When the offset is small it's likely creating a lot of arcs (so the vertex count is higher) when the offset is large it's likely causing many sub regions to collapse (so the vertex count is lower).

There may still be a problem, but the number of vertexes doesn't suggest that there is a problem, you need to visualize it to confirm.

tomicfish, 2021-01-30:
Hi jbuckmccready,
I have done a visual inspection, and the direction and distance of the new curve are correct,  no matter the offset is large or small.
But the number of points of the new curve is far less than that of the original edge when the offset is large.
If I want to get a continuous parallel curve when the offset is large, please give me some suggestions.

jbuckmccready, 2021-01-30:
Can you post some images/visual sketch of what it does now vs. what you're wanting?

tomicfish, 2021-01-30:
The image named "originalImage.png" is the original edge.
The image named "resultImage.png" is the new curve and original edge.
[image]
[image]
I want the points of the new curve to be as continuous as the original curve, rather than intermittently spaced as the new curve is now.

jbuckmccready, 2021-01-30:
From the image it looks like the offset of a single polyline is producing lots of small discontinuous polylines?

jbuckmccready, 2021-01-30:
OK I loaded it up, it does break it into lots of disjoint polylines, might have to do with the density of points causing problems. I'll play with it some more.

tomicfish, 2021-01-30:
Yes, a single polyline produces many small discontinuous polylines when the offset is large,
but a single polyline produces only one continuous polylines when the offset is small.
The image named "resultImage2.png" is the new curve and original edge when the offset is 5.
[image]

jbuckmccready, 2021-01-30:
Scaling it up by 1000 or 10000 does produce less polylines in the offset (so it does seem to help). I don't know what your use case is but if you can use a simple weeding algorithm to reduce the density of points it will probably fix the issue (that would be the approach I would take). It seems way over sampled for the shape, but maybe it's required for the use case.

jbuckmccready, 2021-01-30:
There's not really an easy/simple solution to this type of problem in the algorithm (that I am aware of). It arises due to limited bits per float, using absolute float thresholds for fuzzy comparison (this is needed for consistency with the limited bits per float), and requiring distance checks in the algorithm to remove invalid segments.

If you can reduce the density of points by using a weeding algorithm or approximating sequences of points with arcs this library will work very good.

Another thing to consider is since you don't have any arcs at all the [Clipper library](http://www.angusj.com/delphi/clipper.php) offset algorithm may handle the raw input you posted better.

jbuckmccready, 2021-01-31:
I'm looking at the final output some more, it looks like the slices end points should stitch together (at least when scaled up before offsetting) but they're not, so maybe there is some strange bug at play. Looking into it.

jbuckmccready, 2021-01-31:
I found the bug, it has to do with an assumed order in the slices when it goes to stitch them together which is not guaranteed by an `std::unordered_map`, I'll work on a commit to fix this bug.

tomicfish, 2021-01-31:
I have tested the library you suggested, and it didn't achieve the expected goal.
I entered 3569 sets of data, but only got 1453 sets of data when the offset is set to 50.
So, I'm looking forward to the update of the program.

jbuckmccready, 2021-01-31:
I committed a fix: https://github.com/jbuckmccready/CavalierContours/commit/71d8b8c7762544ec06e2f50beb75b186ea5b415f.

Note for the example input you gave me I still needed to scale it up by 100 to get valid results (this is unavoidable due to limited bits per float). A warning here is that when you scale the output back down it may have very close/overlapping vertexes which will cause problems for the offset algorithm so you'll also need to prune them by calling pruneSingularities if you want to perform offsets repeatedly on the outputs (function for pruning is [here](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polyline.hpp#L440)).

In the future rust port I'll likely have it auto detect and scale internally by default, and allow disabling via a config/option parameter for performance reasons.

tomicfish, 2021-01-31:
I did the test after modifying the code according to the document(71d8b8c), but the result didn't seem to improve.
I also tested the input after scaling it up by 100, but the result still didn't seem to improve.I entered 3569 sets of data, but only got 1365 sets of data when the offset is set to 50.
The image named "noScaleImage.png" is the new curve and original edge when the offset is 50 and no scale the input.
The image named " scale100Image.png" is the new curve and original edge when the offset is 50 and scale input  up by 100.
[image]
[image]

for (size_t i = 0; i < PX.size(); i++)
{
    input.addVertex( (PX[i]*100), (PY[i]*100), 0);
}
input.isClosed() = false;
double offset = 50 *100;
std::vector<cavc::Polyline> results = cavc::parallelOffset(input, offset);
int OutNums = 0;
if(!results.empty())
{
    for (size_t i = 0; j < results.size(); i++)
    {
        OutNums += results[i].size();
        for (size_t j = 0; j < results[i].size(); j++)
        {
	    results[i][j].x() = results[i][j].x()/100;
	    results[i][j].y() = results[i][j].y()/100;
        }
    }
}
This is my code. Is there a problem with my code？

jbuckmccready, 2021-01-31:
You can use the [scalePolyline](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polyline.hpp#L68) function for convenience (but it doesn't do anything special, just multiplies all X and Y values by the scale value). You can surround the code with "```cpp" to have it render better in these GitHub comments (see doc on it [here](https://docs.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax#quoting-code)).

The code you posted looks OK. You keep mentioning and counting the number of vertexes but I don't think that's relevant, the main issue is the number of output polylines and whether it's an accurate constant distance parallel offset (the total number of output vertexes may be higher or lower depending on the type of input and the offset value).

This is the code I'm running that works.

```cpp
cavc::Polyline<double> input;
// polyline defaults to having isClosed false
// fill input polyline with all vertexes by calling addVertex
// ... omitted for brevity

double scale = 100.0;

cavc::scalePolyline(input, scale);
double offset = scale * 50.0;
std::vector<cavc::Polyline<double>> output = parallelOffset(input, offset);

// only one output polyline (the problem before the fix was multiple outputs here)
assert(output.size() == 1);
// scale output back down if wanting to use it with original scale
for (auto &polyline : output) {
  cavc::scalePolyline(polyline, 1.0 / scale);
  // there are no overlapping vertexes in this case of scaling by 100 and offsetting by 50 for this example input
  // but for other values it may have overlapping vertexes after scaling back down
  // uncomment this line if you want to ensure there are no overlapping vertexes after scaling back down the output
  // polyline = cavc::pruneSingularities(polyline, cavc::utils::realPrecisions<double>());
}
```

jbuckmccready, 2021-02-02:
@tomicfish Did you get it working?

tomicfish, 2021-02-04:
Sorry to reply so late, I have done a lot of tests.
After reducing the number of points on the edge, I can get the ideal parallel curve.
Also, I did another test. I can do many operations with an offset of 5 if I want to get a parallel curve with an offset of 50.
But the number of points on the curve grows very fast without reducing the number of edge points. It took me some time to deal with it. If you have any suggestions to get dense continuous edges when the offset is large, please let me know.
Thank you very much for your help. I'm going to close this issue.

closed 2021-02-04 by tomicfish

jbuckmccready, 2021-02-04:
My recommendations for dealing with dense inputs are:
1. For performance and stability use a good weeding algorithm to cut down the number of vertexes (e.g. you can use smallest triangular area between any three consecutive vertexes to determine removing the middle one).
2. For stability scale up the input (as already discussed in this issue).
3. For performance as you already have found it is usually more efficient if there are many offset segments overlapping to perform repeated smaller offsets (the vertex count will grow but it should not continuously grow, vertexes will be added initially to add arcs but after that it will somewhat stabilize).

tomicfish, 2021-02-05:
Thank you for your recommendations！ I will try as you suggest.