jbuckmccready/CavalierContours #35: It doesn't work only offset=0.5 on specific outerpolyline and island.

doomori, 2021-01-30:
Hi,I am doomori.
Thank you very much for wonderful library.

I tried rewrite in interactiveUI/plineoffsetislandsalgorithmview.cpp

cavc::Polyline<double> mainOuterPline;
mainOuterPline.addVertex(0,0,0);
mainOuterPline.addVertex(20,0,0);
mainOuterPline.addVertex(20,20,0);
mainOuterPline.addVertex(0,20,0);
mainOuterPline.isClosed() = true;

and

cavc::Polyline<double> island1;
island1.addVertex(15, 10, 0);
island1.addVertex(17, 10, 0);
island1.addVertex(17, 12, 0);
island1.addVertex(15, 12, 0);
island1.isClosed() = true;
invertDirection(island1);

island is only one,so this code is done on QtCreator 
but,
[image]
this bug?? so it dose not work only offset =0.5,,it work offset =0.4 or offset0.6
I can't understand and resolve,,Please help me.

Thank you .

jbuckmccready, 2021-01-30:
Thanks for very easy to understand and reproducible case. It definitely is a bug. I'm guessing it has to do with all the intersects between the island offset and surrounding offset being overlapping/coincident. I'll try and take a look at how to fix it this weekend.

doomori, 2021-01-30:
Hi,jbuckmccready.Thank you for relay. 
Please fix this bug.
Best Regard.

jbuckmccready, 2021-01-31:
I fixed this issue, commit: https://github.com/jbuckmccready/CavalierContours/commit/159852b9f4481f9547c74af5d11db3ac4e4325d1

Note that it still isn't defined whether to keep coincident segments or not - sometimes the coincident segments remain, sometimes they are removed. I could see wanting this as an option to either keep or discard the coincident segments since both are correct, this is likely something I would tackle in the rust port I am working on.

The bug of not detecting the intersects is fixed and the case you give now works correctly.

Thanks for creating this issue and please create more issues if you find other bugs.

closed 2021-01-31 by jbuckmccready

doomori, 2021-02-01:
I'm done checking the fix.
Thank you for your prompt response!