jbuckmccready/CavalierContours #36: Please tell me the rule(structure) of parentIndex in parallelOffsetIsland.

doomori, 2021-02-09:
Hi,jbuckmccready.I am doomori.
Thank you for good libraly.

Please tell me the rule(structure) of parentIndex in parallelOffsetIsland.
I rewrite that outer Rectangle have 4 simple islands in interactiveUI/plineoffsetislandsalgorithmview.cpp .

mainOuterPline.addVertex(-120,-80,0);
mainOuterPline.addVertex(120,-80,0);
mainOuterPline.addVertex(120,80,0);
mainOuterPline.addVertex(-120,80,0);

and 

int count = 4;
for (int i = 0; i < count; ++i)
  {
      cavc::Polyline<double> island;
      island.addVertex(-100+14*i, 10, 0);
      island.addVertex(-94+14*i, 10, 0);
      island.addVertex(-94+14*i, 15, 0);
      island.addVertex(-100+14*i, 15, 0);
      island.isClosed() = true;
      invertDirection(island);
       m_cwLoops.push_back(std::move(island));
}

and 
(in PlineOffsetIslandsAlgorithmView::updatePaintNode)
for (auto const &loop : loopSet.cwLoops) {
        printf("cw_parent_Index%zu\n",loop.parentLoopIndex); <-add
          addPline(loop.polyline);
      }
for (auto const &loop : loopSet.ccwLoops)
{
    printf("ccw_parent_Index%zu\n",loop.parentLoopIndex);<-add
  addPline(loop.polyline);
}

offset = 2.5
The result is

cw_parent_Index1
cw_parent_Index2
cw_parent_Index3
cw_parent_Index4
ccw_parent_Index0
------------------
cw_parent_Index0
ccw_parent_Index0
------------------
cw_parent_Index1
ccw_parent_Index0
------------------
cw_parent_Index0
ccw_parent_Index0
------------------
ccw_parent_Index0
------------------
ccw_parent_Index0

In my sence,

cw_parent_Index1 -> 0
cw_parent_Index2 -> 1
cw_parent_Index3 -> 2
cw_parent_Index4 -> 3
ccw_parent_Index0
------------------
cw_parent_Index0  any of 0-3
ccw_parent_Index0
------------------
cw_parent_Index1 -> 0 
ccw_parent_Index0
------------------
cw_parent_Index0
ccw_parent_Index0
------------------
ccw_parent_Index0
------------------
ccw_parent_Index0

Please tell me the rule (structure) of parentIndex about cwLoops.
Thank you .

jbuckmccready, 2021-02-10:
They are currently only used internally so there isn't any structure to them externally (probably shouldn't even be part of the API). In the future it would be nice to return the list of all parent indexes (it may be more than 1) associated with each resulting loop.

doomori, 2021-02-10:
I understood.Thank you for answering!

closed 2021-03-03 by jbuckmccready