jbuckmccready/CavalierContours #38: (Question) Smooth offset contour

apeltauer, 2021-03-19:
Hi and many thanks for this great library.
I would like to try to generate a smooth offset contour, with at best only tangential connections.
The offset contour on an rectangle island is already a good result. I know this comes from the fact that the offset lines are connected by an arc. The same could be also done by all of the others too. 
So my question would be, could you point me a place where the arc calculation is done and where the "smooth" algorithm could be placed?

BR Manuel

jbuckmccready, 2021-03-19:
I do not quite follow what you mean by a "smooth offset contour, with at best only tangential connections" (I'm curious to see an example picture).

For every polyline segment (two adjacent vertexes) the algorithm generates a raw offset segment which is the correct constant parallel offset for that single segment. Then for each pair of raw offset segments that neighbor each other they may either intersect each other (in which case it trims/clips both segments to the intersect point) or be disjoint (in which case the segment end points are connected with an arc).

Here is a picture (blue lines are the input polyline, purple segments are the raw offset segments, red circles are the input polyline vertexes):
[image]

After all the raw offset segments have been joined/trimmed together they form a raw offset polyline. The raw offset polyline is then sliced at all of its self intersects, and all slices which are too close (distance smaller than the parallel offset value) to the input polyline are discarded. The remaining slices are all stitched together end to end to form the final output results as polylines.

If what you want to do leads to offset segments being closer to the input polyline than the offset distance it will require changing quite a bit of code because the slice discard step requires that the raw offset polyline segments be no closer to the input polyline than the offset distance.

jbuckmccready, 2021-03-19:
It may be easiest to just have a second step in processing after performing the offset you could walk the segments and perform the "smoothing" operation you intend.

apeltauer, 2021-03-20:
Sorry for not providing a picture :)

I would like to also connect the already intersection offset segments by an arc.
Can you please point me to the code snip where you create/calculate the arc for the not intersection segments? 
From there i may  find a solution to calculate an arc. 
For me this is the first time working with polyline and the bulg value :)

[image]

jbuckmccready, 2021-03-20:
So from the picture it looks like whenever the segments intersect you want to form an arc join between them rather than trim to the intersect point? I think that should still work with the rest of the algorithm and still pass the distance check it performs when discarding slices.

These are the functions that join/trim the raw offset segments:
[lineToLineJoin](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polylineoffset.hpp#L108)
[lineToArcJoin](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polylineoffset.hpp#L157)
[arcToLineJoin](https://github.com/jbuckmccready/CavalierContours/blob/master/include/cavc/polylineoffset.hpp#L228)

These functions find the intersect between two adjacent raw offset segments (lineline, linearc, or arcarc) and then either trim or connect them using an arc based on the intersect result.

apeltauer, 2021-03-22:
Thanks for your reply. I will try out those functions...
Many thanks...