jbuckmccready/CavalierContours #43: can you explain the parameters

imgtech0, 2021-08-07:
Hello JBuck and thanks for your excellent work!
I am reading the code, but  i have a question that that how to assign the value  to the parameters currSliceIndex and available  int the folowing  code.
avc\polylinecombine.hpp
  auto createUnionAndIntersectStitchSelector = [](std::size_t startOfCoincidentSlicesIdx) {
    return [=](std::size_t currSliceIndex, std::vector<std::size_t> const &available)

Thanks

jbuckmccready, 2021-08-07:
The stitch selector functions are used to resolve the case where there are multiple slices that could be stitched together at the same point in 2D space. `currSliceIndex` is used to determine what type of slice is being stitched (slice index is used to compare with `startOfCoincidentSlicesIdx` to determine if the slice is a coincident one or not) and `available` is all the possible slices (as indexes) that could be stitched to the current one. The slice indexes are referring to the index position of the slice in the vector containing all the slices, you can see that the the vector of slices and stitch selector are passed into the `stitchOrderedSlicesIntoClosedPolylines` function.

Sorry it's not more documented, I'd like to build out the documentation more and clean up the code in the continued Rust project here: https://github.com/jbuckmccready/cavalier_contours

imgtech0, 2021-08-21:
thank you very much.
I want to compile the project in visual stdio 2013,but it dose not support  c++14.  I have a question that could you modify this project,  and make it can be complied in  visual stdio 2013 environment?  I you can do this Please tell me the time used and how much does it cost?

Thanks

imgtech0, 2021-08-21:
If u can do this, u can connect me by email: imgtech@126.com
thank u