jbuckmccready/CavalierContours #44: How to make a contour with sharp corners instead of arcs?

lenochod, 2021-09-01:
Hello,

thanks for this highly useful library. Having used it to generate contours for a while to great satisfaction, we found that we need to avoid the rounded corners of contours in some special applications and need sharp angles instead. We are using the `cavc::parallelOffset()` function to generate contours. 

Could you please advise us on where to start modifying the code? I started with `case LineSeg2LineSeg2IntrType::False` inside `lineToLineJoin()` which helped me a lot but I'm not sure if this is the right place and may be there are other places to check as well.

Thanks

[image]

timoria21, 2023-05-24:
Hello lenochod,

Did you succeed with the approach you suggested? Did you encounter some issues? 
We now need to do the same thing and it would be nice if you can share some code. 

Thank you.

lenochod, 2023-05-25:
Hi timoria21,

the business need for it ceased on our side so we did not continue. I don't have code to share and I don't remember much of the details after almost 2 years.

Best regards,

lenochod

timoria21, 2023-06-07:
Thanks lenochod,

I was able to implement a two-lines connector to cover this case.

Thanks again for answering.

Anruin, 2024-06-19:
[image]

We can calculate the direction of the segment 1 using its v2 and v1 difference, then use the line segment intersection parameter t0 to extend the s1.v1 position in the direction of the segment. Don't normalize it as we will lose length and distance will be applied incorrectly.