jbuckmccready/CavalierContours #50: 3D Offset on surface

kadircanik, 2022-03-09:
Do you knowledge about 3d offseting a curve or line on a surface?

jbuckmccready, 2022-03-09:
I don't have much. I didn't do any searching/research on it, here's some thoughts just off the top of my head (could be totally wrong): I think you can turn the problem into a 2D problem. You need transformation functions to project to 2D and back to 3D, then you can project to 2D, perform the offset, and project back to 3D. The transformation functions are determined by what direction you want to offset relative to the surface, so the surface is really just acting as a "guide".

Imagine you had a curve going through 3D space (not restricted to a surface) and you wanted to offset it - first you need to determine what direction you're offsetting, there are infinite possibilities. The direction can be set by projecting the curve to a 2D plane and then the offset operation can take place in 2D, then the offset result can be projected back to 3D using the projection plane that was used to go to 2D.

kadircanik, 2022-03-09:
Thank you for answer.
Actually your suggestion will not give correct result.
2D offsetting guaranties the new curve is equidistant but after projecting to the support 3d surface it will not be equidistant.
Commercial cad softwares does it well.
I am looking for an opensource alternative.
Anyway thanks again for your time and answer.
Best regards.

closed 2022-03-09 by kadircanik