jbuckmccready/CavalierContours #53: Contours of shifted polylines differ

lenochod, 2022-06-04:
Hello,

first, thanks for the great library. We've been using it to our satisfaction for quite some time. However, we are facing an issue which looks like numeric instability to me. We will appreciate your help. Thanks.

We have a drawing with multiple figures like this (ignore the dashed lines):
[image]

We are generating both inner and outer contour for it. It works fine on this closed polyline

```
X	Y	Bulge
473.760000000000000	123.560000000000000	0.463985680061077
475.033239959260000	138.783314391340000	0.463985680061077
459.809938967064000	137.509914237247000	0.134441584771250
444.227406100013000	140.395488288632000	0.041688391650421
439.303094239882000	139.562816356986000	0.273832882708070
417.687160318691000	117.946814732389000	0.041688391650421
416.854503810494000	113.022500264269000	0.134440937017869
419.740000000000000	97.440000000000000	0.463985680061078
418.466760040740000	82.216685608660400	0.463982482960671
433.690000000000000	83.490000000000000	0.134441734196030
449.272593899987000	80.604511711367700	0.041688391650421
454.196905760118000	81.437183643013500	0.273832882708070
475.812839681309000	103.053185267611000	0.041688391650421
476.645496189506000	107.977499735731000	0.134441734196029
473.759959094621000	123.560084597999000	0.000000000000000
```

But it fails to generate the inner contour on this closed polyline which differs just by adding 248 to all the X coordinates. The remaining diffs are <~1e-12. 
```
X	Y	Bulge
721.760000000000000	123.560000000000000	0.463985680061079
723.033239959260000	138.783314391339000	0.463985680061078
707.809938967064000	137.509914237247000	0.134441584771250
692.227406100013000	140.395488288632000	0.041688391650421
687.303094239882000	139.562816356986000	0.273832882708070
665.687160318691000	117.946814732389000	0.041688391650422
664.854503810494000	113.022500264269000	0.134440937017869
667.740000000000000	97.440000000000000	0.463985680061077
666.466760040740000	82.216685608660500	0.463982482960672
681.690000000000000	83.490000000000000	0.134441734196030
697.272593899987000	80.604511711367700	0.041688391650421
702.196905760118000	81.437183643013500	0.273832882708070
723.812839681309000	103.053185267611000	0.041688391650421
724.645496189506000	107.977499735731000	0.134441734196029
721.759959094621000	123.560084597999000	0.000000000000000
```

We worked around this problem by shifting the polyline to [0,0], generate contours and then shift back. Here's the data when shifted to [0,0]
```
X	Y	Bulge
0.000000000000000	0.000000000000000	0.463985680061079
1.273239959259740	15.223314391339500	0.463985680061078
-13.950061032935700	13.949914237247200	0.134441584771250
-29.532593899986500	16.835488288632300	0.041688391650421
-34.456905760118000	16.002816356986500	0.273832882708070
-56.072839681309000	-5.613185267610900	0.041688391650422
-56.905496189505900	-10.537499735731300	0.134440937017869
-54.020000000000000	-26.120000000000000	0.463985680061077
-55.293239959259800	-41.343314391339500	0.463982482960672
-40.070000000000100	-40.070000000000000	0.134441734196030
-24.487406100013500	-42.955488288632300	0.041688391650421
-19.563094239881900	-42.122816356986500	0.273832882708070
2.052839681309020	-20.506814732389100	0.041688391650421
2.885496189505940	-15.582500264268700	0.134441734196029
-0.000040905378796	0.000084597999020	0.000000000000000
```

Now it generates a contour but it is not closed - see this picture, ignore the dashed lines:
[image]

For your reference, below is the code we used. May be there's something wrong in the way we call the library. We multiply all the coordinates and offset by 10000 prior to calling this function because of previous issues where some contours were not generated completely. And we divide by 10000 after contours are generated. Offset is 0.36 prior to scaling up.

```C++
vector<Polyline<double>> CreateContours(Polyline<double> &input, double offset, vector<PlineIntersect<double>>& selfIntersects) {
  // Prune the input polyline of singularities.
  Polyline<double> pruned = pruneSingularities(input, SAME_POINT_EPSILON);

  // Detect self-intersects in the pruned polyline
  selfIntersects.clear();
  if (pruned.size() >= 2) {
    // Spatial index requires at least two points,
    // otherwise the program aborts.
    auto spatialindex = createApproxSpatialIndex(pruned);
    allSelfIntersects(pruned, selfIntersects, spatialindex);
  }

  // compute the resulting offset polylines
  if (selfIntersects.empty()) {
    auto contours = parallelOffset(pruned, offset, false);
    if (contours.size() == 0) {
        // parallelOffset(pruned, offset, true) sometimes does not generate
        // contours due to numeric errors. In such a case, try 
        // parallelOffset(pruned, offset, false) which takes a different code 
        // path and more time but succeeds.
      contours = parallelOffset(pruned, offset, true);
    }
    return contours;
  } else {
    return parallelOffset(pruned, offset, true);
  }
}
```

jbuckmccready, 2022-06-04:
Hmm, this is due to one or more of the bugs I fixed in the Rust code since it works correctly using the [Rust version](https://github.com/jbuckmccready/cavalier_contours).

I tested it using the interactive web demo [here](https://jbuckmccready.github.io/cavalier_contours_web_demo_page/), you can paste in the following for the polyline under the "Polyline Offset" tab (this is the polyline shifted to [0,0] but I tested with the other lines you posted as well):
```json
{
  "isClosed": true, 
  "vertexes": [
    [0.000000000000000, 0.000000000000000, 0.463985680061079],
    [1.273239959259740, 15.223314391339500, 0.463985680061078],
    [-13.950061032935700, 13.949914237247200, 0.134441584771250],
    [-29.532593899986500, 16.835488288632300, 0.041688391650421],
    [-34.456905760118000, 16.002816356986500, 0.273832882708070],
    [-56.072839681309000, -5.613185267610900, 0.041688391650422],
    [-56.905496189505900, -10.537499735731300, 0.134440937017869],
    [-54.020000000000000, -26.120000000000000, 0.463985680061077],
    [-55.293239959259800, -41.343314391339500, 0.463982482960672],
    [-40.070000000000100, -40.070000000000000, 0.134441734196030],
    [-24.487406100013500, -42.955488288632300, 0.041688391650421],
    [-19.563094239881900, -42.122816356986500, 0.273832882708070],
    [2.052839681309020, -20.506814732389100, 0.041688391650421],
    [2.885496189505940, -15.582500264268700, 0.134441734196029],
    [-0.000040905378796, 0.000084597999020, 0.000000000000000]
  ]
}
```
[image]

I have not been actively maintaining the C++ since moving to Rust - it would be nice to update the C++ to work the same as the Rust code but it's a lot to maintain and I just wont be able to find the time/motivation to do it. There is a C API wrapper to the Rust code which you can use from C++ which is what I recommend. The Rust code is as fast (or faster due to avoiding some allocations and using a better map container) and written entirely in safe code to eliminate any memory bugs.

If you'd like to fix it in C++/maintain the C++ I'm open to pull requests and I can maybe help steer where to look. The issue is likely due to changes in handling of intersects or in the way the slices are stitched together - you have to step through the code using the non-working example (I recommend eliminating vertexes to reduce it to minimal non-working example) and see where it fails to stitch slices together or fails to find an intersect, etc. The functions/types are structured mostly the same between Rust and C++ so you could also step through/print out the results of stages from the Rust and compare with the C++ to isolate the area of the code.

If you have other ideas/feedback or questions feel free to reply to this issue.

lenochod, 2022-06-10:
Hi, 

thanks for your prompt answer and sorry for a late reply. I somehow missed the email notification.

I completely understand that you want to avoid dual maintenance and I will upgrade our solution to the Rust version of the library.

jbuckmccready, 2022-06-10:
If you have any suggestions/questions/problems upgrading to Rust let me know.

lenochod, 2022-06-14:
Thanks. I was able to convert our C++/CLI wrapper to use the Rust library and I'm testing it now. If you want, I can describe the process on the Wiki or may be even share the wrapper. In general, the process was much easier than I expected.

Just a minor wish: could you please add a build step to automatically generate the C header file? I used cbindgen, not a big problem, but getting both the DLL+LIB and header by the same build would be nice. Thanks.

jbuckmccready, 2022-06-14:
Nice, glad you got it working!

`cbindgen` is kind of heavy to build so I'm hesitant to add to the Rust build script `build.rs`. But there could be clear instructions on how to install and run `cbindgen` using `cargo`, the instructions could be in the `README.md` in the `cavalier_contours_ffi` crate directory, then it can be linked to from the root level `README.md`. Also the readme can link to the `cbindgen` user guide [here](https://github.com/eqrion/cbindgen/blob/master/docs.md).

Any thoughts/opinions? I want to make it easy to use but I am also trying to avoid bloating the repository with specific use case build scripts/scenarios.

I'm currently using it from C# so I didn't actually have a use for the C header but I could see it being helpful for documentation purposes, or for use as an input to another language binding generator.

lenochod, 2022-06-14:
Clear instructions on how to use `cbindgen` would help. But it was a simple google search to get it, not a big problem.

More interesting to me is that you use it with C# and did not need the C/C++ header files. How are you doing this? The project builds just a single DLL, `cavalier_contours_ffi.dll` and it's not a .NET assembly. We were using the C++ version in C# through a Visual C++ project with CLR support which added the .NET wrapper. So I did the same with `cavalier_contours_ffi.dll` and that's why I needed the header file. But it would be much better if we could use the lib directly without such a wrapper.

jbuckmccready, 2022-06-14:
I just use [P/Invoke](https://docs.microsoft.com/en-us/dotnet/standard/native-interop/pinvoke).
Looks something like this in C#:
```csharp
[StructLayout(LayoutKind.Sequential)]
public struct PlineVertex
{
    public double X;
    public double Y;
    public double Bulge;
    public PlineVertex(double x, double y, double bulge)
    {
        X = x;
        Y = y;
        Bulge = bulge;
    }
}

public static class NativeMethods
{
    private const string LibFileName = "path_cooker_ffi.dll";

    [DllImport(LibFileName, CallingConvention = CallingConvention.Cdecl)]
    public static extern int cavc_pline_create(PlineVertex[] vertexes, uint n_vertexes, byte is_closed, out IntPtr pline);

    [DllImport(LibFileName, CallingConvention = CallingConvention.Cdecl)]
    public static extern void cavc_pline_f(IntPtr pline);

    [DllImport(LibFileName, CallingConvention = CallingConvention.Cdecl)]
    public static extern int cavc_pline_add(IntPtr pline, double x, double y, double bulge);
}
```

So all the functions declarations are in CSharp (method names match the `extern "C"` functions from the Rust C FFI). Then I create a C# wrapper polyline class which calls `NativeMethods.cavc_pline_create` in the constructor (throws exception if non-zero return) and calls `NativeMethods.cavc_pline_f` in its `IDisposable.Dispose` method.

Sorry I can't share the code, it's not open source, but hopefully this reply helps.