jbuckmccready/CavalierContours #54: bulge value error when bulge>1?

airsxue, 2022-06-22:
Hi,
Thanks for your great work.
Just want to check why the code will show erree when my input |bulge| >1? and when I split this arc to 2 arcs the code works well.Could you please kindly tell me the reason why the |bulge| cant bigger than 1? and what should i can do to deal with the situation when a arc's |bulge|>1? Thanks a lot.
[image]

jbuckmccready, 2022-06-22:
The algorithm only processes segments which are lines or arcs that are no more than half circles/180 degrees. If you want to work with larger bulge values you can split the arc into two (as you have shown).

If you want the output to have larger bulge values then you can stitch the arcs back together after processing by finding consecutive arcs which are tangent. You can use [this](https://github.com/jbuckmccready/CavalierContours/blob/85766228c9843326ce59ce0577b4793a7e77f306/include/cavc/plinesegment.hpp#L133-L148) function to find the tangent vector (not normalized) at the end points of the arc segments. Then you can normalize the tangent vectors and compare for equality (using some epsilon value for fuzzy float comparing).

airsxue, 2022-06-22:
> 

Well got. Many thanks for your explanation。