jbuckmccready/CavalierContours #60: generate offset contour of a line segment

kalluwa, 2023-04-10:
i tested a line segment in rust web demo, it works fine:
[image]

but this cpp demo failed:
[image]

after adding a another point,it can produce right result:
[image]

could this be normal behavior?

jbuckmccready, 2023-04-11:
This is likely a bug in the C++ code (since there are many that have been fixed in Rust and not back ported), I am not maintaining the C++ code, see this pinned issue: https://github.com/jbuckmccready/CavalierContours/issues/41

kalluwa, 2023-04-11:
thanks for your reply. 
it should not be a difficult problem(i guess), i'll debug it later

closed 2023-04-11 by kalluwa