jbuckmccready/CavalierContours #62: New incorrect case

timoria21, 2023-07-24:
```
// input polyline
cavc::Polyline<double> input;
// add vertexes as (x, y, bulge)
input.addVertex(-30.425, -19.3125, 0);
input.addVertex(-30.425, -19.6875, 0);
input.addVertex(45.575, -19.6875, 0);
input.addVertex(45.575, -19.3125, 0);
input.addVertex(102.575, -19.3125, 0);
input.addVertex(102.575, 77.25, 0);
input.addVertex(-87.3, 77.25, 0);
input.addVertex(-87.3, -18.9375, 0);
input.addVertex(-30.425, -18.9375, 0);
input.isClosed() = true;

cavc::Polyline<double> b;
// add vertexes as (x, y, bulge)
b.addVertex(45.575, -19.3125, 0);
b.addVertex(102.575, -19.3125, 0);
b.addVertex(102.575, -18.9375, 0);
b.addVertex(45.575, -18.9375, 0);
b.isClosed() = true;

cavc::CombineResult<double> res = cavc::combinePolylines(input, b, cavc::PlineCombineMode::Exclude); // As expected

cavc::Polyline<double> pA2;
pA2.addVertex(-48.583333333333364, -32.104166666666657, 0);
pA2.addVertex(-48.583333333333364, -32.479166666666657, 0);
pA2.addVertex(27.416666666666686, -32.479166666666657, 0);
pA2.addVertex(27.416666666666686, -32.104166666666657, 0);
pA2.addVertex(84.416666666666686, -32.104166666666657, 0);
pA2.addVertex(84.416666666666657, 64.458333333333343, 0);
pA2.addVertex(-105.45833333333334, 64.458333333333343, 0);
pA2.addVertex(-105.45833333333331, -31.729166666666664, 0);
pA2.addVertex(-48.583333333333314, -31.729166666666664, 0);
pA2.isClosed() = true;

cavc::Polyline<double> pB2;
pB2.addVertex(27.416666666666686, -32.104166666666664, 0);
pB2.addVertex(84.416666666666686, -32.104166666666664, 0);
pB2.addVertex(84.416666666666686, -31.729166666666664, 0);
pB2.addVertex(27.416666666666686, -31.729166666666664, 0);
pB2.isClosed() = true;

cavc::CombineResult<double> res2 = cavc::combinePolylines(pA2, pB2, cavc::PlineCombineMode::Exclude); // Not as expected
```
[image]

jbuckmccready, 2023-07-27:
This is fixed in the Rust code, try putting those polylines in to the interactive demo here: https://jbuckmccready.github.io/cavalier_contours_web_demo_page/#/pline_boolean

timoria21, 2023-07-27:
Can you help me to find the changeset that resolves this issue? I'm currently unable to switch to Rust code.

Thank you.

jbuckmccready, 2023-07-27:
I have made a lot of fixes so I really don't know which would apply, best bet if you must use the C++ code is to follow the code in both projects and track down what has changed (that is what I would do - or re-implement the C++ code according to the Rust code). If you'd like to help fix it/maintain the C++ side of things I can accept pull requests or also look into making you a maintainer of the repo so you can commit to it directly.

I made a lot of small fixes with the intersect handling (the primitive segment intersect functions for line-line, line-arc, and arc-arc), so that would probably be the easiest place to start as those functions are small. But I know I made other changes as well for finding all intersects between two polylines and the boolean operation code.

Note you can both dynamically and statically link to the Rust C FFI from C++ so there should be no reason you can't use it other than some sort of company policy.