jbuckmccready/CavalierContours #64: Self-intersection

LV6699, 2024-01-09:
If polyline has self-intersection, is there any theoretical guidance to deal with the self-intersection problem?

jbuckmccready, 2024-01-09:
Self intersection can be handled by enabling it for offsetting single polylines. For multiple/island offsetting and boolean operations it's not supported - I don't know of a good work around for the way the algorithms currently work, probably work needs to be done to improve the core algorithms.

LV6699, 2024-01-11:
First of all, thank you for your wonderful answer. For this algorithm, if the input is a single non self intersecting polyline curves, after the algorithm's self intersection calculation, the output may have multiple non self intersecting loops. Among these loops, some are valid loops, while others are invalid loops. How to determine whether a closed non self intersecting loop is valid? Or how do we determine if a closed non self intersecting  loop is what we want? How does this algorithm solve this problem?

jbuckmccready, 2024-01-11:
The algorithm works by performing a distance check from the mid point of each segment of the resulting offset polyline to the original polyline (using a spatial index to avoid checking against every original polyline segment). The distance is calculated and if it is less than the offset value plus some small epsilon then the entire polyline the segment midpoint belongs to is discarded.